rm-hull/luma.led_matrix #155: Doesn't work with Dragonboard 410c

lmsb, 2018-07-22:
The display is all garbled. Library and python code examples run fine but actual LEDs display garbled.

rm-hull, 2018-07-22:
Can you post details of:
* What command or program you ran
* How many LED matrix units you have, and how you have them wired up
* A picture of the garbled LEDs
* A url link to to the device you have purchased

rm-hull, 2018-07-23:
ok, things I would do:
* Remove all the wires, and run an SPI sanity check - see https://github.com/rm-hull/spidev-test
* remove any other wires from the board other than those needed for driving the LEDs
* check that there are no other programs/processes running that also use SPI
* check/double-check my wiring between the board and the LED matrix

lmsb, 2018-07-25:
@rm-hull , 
I'm now suspecting that the problem is speed. 
I say this because I followed your advise and disconnected everything else not related with the SPI lines, then ran the matrix_demo and still no activity. Then, went and lowered the speed on the spi interface at the matrix_demo.py program and I could see the display with some activity. I tried to reduce the speed further but there's an assert under the "luma.core.interface.serial" library file which doesn't allow me to specify a value of Hz lower than 500,000. 
**Question**: Can I further reduce the speed here? how? 
Suspected about the speed first because I'm using a voltage level shifter because the SPI in DB410c outputs at 1.8v while the max7219 display works at 5v. And based on this [forum entry ](https://forum.digilentinc.com/topic/8781-pmod-lvlshft/?page=0#comment-26133) I found that we must use a lower speed when shifting from/to 1.8v. 

I checked the specs on the DB410c board and the Raspberry pi and the DB410c has a higher clock speed than the Rpi2 but has the same as the Rpi3. But even so, according to this [link](https://towardsdatascience.com/rpi3-vs-dragonboard-f4dd877b7da9), the DB410c still remains faster than the Rpi3 under different performance tests. 

Another test I did was to use the spidev_test utility and all looked ok as after bridging the MOSI and MISO with a wire I was able to confirm the looped data. 

Thanks again for your help.

lmsb, 2018-07-25:
@rm-hull , 

Would you have a basic set of byte commands to be used with the spidev_test to drive the max7219 LED display ? 

thanks,
Luis.