rm-hull/luma.led_matrix #157: Add degree symbol to charmap

petrkr, 2018-07-31:
Add degree symbol to charmap, so we can use this library to show temperature.

```
diff --git a/luma/led_matrix/segment_mapper.py b/luma/led_matrix/segment_mapper.py
index 91c3447..d6a8e5f 100644
--- a/luma/led_matrix/segment_mapper.py
+++ b/luma/led_matrix/segment_mapper.py
@@ -6,6 +6,7 @@ _DIGITS = {
     ' ': 0x00,
     '-': 0x01,
     '_': 0x08,
+    '°': 0x63,
     '\'': 0x02,
     '0': 0x7e,
     '1': 0x30,
```

[image]

rm-hull, 2018-08-01:
Seems useful .. I will create a PR at the weekend unless @petrkr fancies doing that?

petrkr, 2018-08-01:
it's ok, I do not see point on doing whole repo clone just to add 1 line in 1 file. Just add it due normal coding.

Petr

thijstriemstra, 2018-09-08:
> it's ok, I do not see point on doing whole repo clone just to add 1 line in 1 file. Just add it due normal coding.

Luckily we have tests that pointed out this doesn't work on Python 2.7. Not as simple as adding 1 line in 1 file unfortunately.. What version of Python were you using @petrkr?

petrkr, 2018-09-08:
That laptop where I tested it got stolen. But everything I do on Python 3.x. Python 2 is already obsolete, does not make sense to use it or even do new things above it.

thijstriemstra, 2018-09-08:
Sorry to hear that!

> Python 2 is already obsolete, does not make sense to use it or even do new things above it.

PSF still supports it, and so do we.

petrkr, 2018-09-08:
Then you can use probably some \codex version of degree symbol, as Python 2 is NOT utf-8 and only ASCII based, there could be that problem. But I do not use it for many years, so do not remember already.

commit fb5352a5 (2018-08-03): Add degree symbol to charmap

fixes #157
PR #158 (2018-09-18): Add degree symbol to charmap
Fixes #157
commit 3af2555d (2018-09-18): Add degree symbol to charmap (#158)

Fixes #157
* Updated gitignore to cover pytest cache files
* Add degree symbol to charmap
* try unicode in py27
* revert prev change
* Rework segment mapper implementation after unicode fix in luma.core
* Add @petrkr to contributors list
* Additional tests
closed 2018-09-18 by rm-hull