rm-hull/luma.led_matrix #159: getting display() to work properly

polterbyte, 2018-09-08:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

Raspberry Pi 3 B Plus

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

I am trying to use display() to work, showing a (rather crude) drawing of a pair of eye on two daisy chained 8x8 matrices. The setup works fine with all examples provided, and I can tweak them to chance text, speeds and other characteristics of the displayed content.

Still, I can't get display() to work, that is, I can't display drawings of my own concoction.

Here is the code that got me the farthest: it doesn't generate errors and <<something>> (I can't be sure what) flashes briefly on the displays. The code:

import numpy as np
from PIL import Image

from luma.core.interface.serial import spi, noop
from luma.led_matrix.device import max7219

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=2)

pixels = np.array([[000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000],
				   [000,000,000,255,255,000,000,000,000,000,000,255,255,000,000,000],
				   [000,000,255,000,000,255,000,000,000,000,255,000,000,255,000,000],
				   [000,255,000,000,000,000,255,000,000,255,000,000,000,000,255,000],
				   [000,255,000,000,255,000,255,000,000,255,000,000,255,000,255,000],
				   [000,255,000,255,255,000,255,000,000,255,000,255,255,000,255,000],
				   [000,000,255,000,000,000,255,000,000,255,000,000,000,255,000,000],
				   [000,000,000,255,255,255,000,000,000,000,255,255,255,000,000,000]])
				   
img = Image.fromarray(pixels, '1')

device.display(img)

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

Something (I can't be sure what) flashes briefly across the displays. And that's it. The code runs without errors, but nothing happens.

I have tried putting the display() command on an infinite loop, but still no result.

If I could be so bold, can an example of the correct usage be provided?

Thanks for any support you can provide (I am quite new on both python, raspberry pi and luma).

polterbyte, 2018-09-08:
Thanks for the prompt reply, Richard.

I ran the code you proposed. The cascaded displays blink for a fraction of a second (not even time enough to identify what is being plotted), and then just disappears. 

Incidentally, I had tried time.sleep() before, with the same results.

Could this be a bug on display()?

polterbyte, 2018-09-08:
The examples work fine, and I even got an example you provided here on issues, that lights up random dots, four at a time.

polterbyte, 2018-09-08:
Oh, and there’s nothing else running on the pi.

Ombruz, 2018-09-12:
Same behaviour of my case (posted on "display output issue"), to me even the examples are not working at all, and nothing else running on the pi.

rm-hull, 2018-09-13:
Would recommend you run an SPI loopback test to isolate whether there is extraneous chatter on the SPI bus - see https://github.com/rm-hull/spidev-test

thijstriemstra, 2019-03-18:
No feedback, closing.

closed 2019-03-18 by thijstriemstra