rm-hull/luma.led_matrix #160: Display output issue

Ombruz, 2018-09-09:
Hello, I installed all the required drivers, I'm running a 4in1 max7219 display connected to a Raspberry Pi zero W and raspian os.
The display is connected to a separated external 5V power supply.
Python is the latest version, and I had no errors while installing or building dependancies.
While running example scripts to test the display output or any kind of script (clocks, messages etc) the output from the display results in fast flickering bogus symbols, solids squares and random led turning on and off in fast sequence.
I tried to reinstall several times, as well as changing the display, the result is still the same.
Any idea about a workaround? Thank you vey much.

Linux raspberrypi 4.14.62+ #1134 BST 2018 armv6l GNU/Linux

Ombruz, 2018-09-10:
I tried either 5v and 3v (including solely from the PI rail), either from the PI and external power sources, still result even with a common ground.
I made a video to show you the display behaviour: 
https://www.youtube.com/watch?v=aGjg98i2fpg&feature=youtu.be

Ombruz, 2018-09-17:
Good evening,
As suggested I run the spidev_test, and this is the output:
[image]
[image]
According to the output seems there is no RX altough every new instance of the test shows different random static patterns on the display as you can see on the second pic.
I checked the correct connection between the MOSI/SCLK pins, also tried those on SPI1 (pins 20 and 21).
Apparently nothing has changed.
Any idea?
Thank you in advance!

rm-hull, 2018-09-17:
Could you take a picture showing the wires going out of the RPi header onto the LED matrix pins?

Can you supply a URL/link where you bought the LED matrices please? Did you solder them yourself or did they come preassembled?

For the SPI test, please remove all / any external wires from the GPIO headers, and as per https://github.com/rm-hull/spidev-test#testing-the-spi-bus,  connect pin 19 (BCM 10 = MOSI) to pin 21 (BCM 9 = MISO) to form a loopback connection. If you aren't seeing the same bytes be transmitted as received, then this would be something that needs addressing before we can investigate the LED matrices.

Ombruz, 2018-09-19:
Good evening
These are the connection I made between the two MAX7219 4in1 matrix:
[image]

These are the cables from the display:
[image]

And these from the raspi side:
[image]

The test using the loopback connection in the way you suggested resulted satisfactory, reading the same bytes on TX and RX.

The unit come from an online shop called Taobao (a sort of aliexpress) but I can't recall exactly the vendor. I just soldered the two blocks made of 4 display each.

rm-hull, 2018-09-19:
@Ombruz - please refer to https://luma-led-matrix.readthedocs.io/en/latest/install.html#max7219-devices-spi

Ombruz, 2018-09-19:
You're great! I changed the order of wires this way:
DIN on MOSI pin
CLK on CE0
CS on SCLK
And I have the first success:
https://www.youtube.com/watch?v=aLpn27vG3r0&feature=youtu.be

closed 2018-09-26 by thijstriemstra