rm-hull/luma.led_matrix #161: UnicodeWarning in `segment_mapper.py`

dentex, 2018-09-25:
#### Type of Raspberry Pi
RPi3.

#### Linux Kernel version
Linux multimedia-pi 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST 2018 armv7l GNU/Linux

#### Expected behaviour
When running this example (or any implementation of mine, for what is worth), I get this warning:

    pi@multimedia-pi:~$ python GIT/luma.led_matrix/examples/matrix_demo.py 
    /usr/local/lib/python2.7/dist-packages/luma/led_matrix/segment_mapper.py:10: 
    UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - 
    interpreting them as being unequal
    '\xb0': 0x63,

Surely it's harmless (and goes away with the `-W ignore` argument), but it's just to let you know.

Thanks, bye.

thijstriemstra, 2018-09-25:
related to #158

dentex, 2018-09-25:
Yes: changing that line to

    u'\xb0': 0x63,

removes the warning.

thijstriemstra, 2018-09-25:
Weird, made the change but breaks the tests: https://travis-ci.org/rm-hull/luma.led_matrix/builds/433082692?utm_source=github_status&utm_medium=notification

rm-hull, 2018-09-26:
v1.1.1 released to pypi with a fix for this issue