rm-hull/luma.led_matrix #163: led_matrix/segment_mapper.py:10:

viktor6, 2018-09-26:
Help
```
pi@raspberry:~/max7219 $ python examples/matrix_demo.py --cascaded 4 --block-orientation -90
/usr/local/lib/python2.7/dist-packages/luma/led_matrix/segment_mapper.py:10: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  '\xb0': 0x63,
Created device
MAX7219 LED Matrix Demo
```
sudo usermod -a -G spi,gpio pi
sudo apt-get install build-essential python-dev python-pip libfreetype6-dev libjpeg-dev
sudo pip install max7219
git clone https://github.com/rm-hull/max7219.git
cd max7219
sudo pip install -e .
sudo pip install spidev
sudo python setup.py instal

rm-hull, 2018-09-26:
See #161 & #162 .. it is a warning so the code should still work as normal

Closing as duplicate

closed 2018-09-26 by rm-hull

rm-hull, 2018-09-26:
v1.1.1 released to pypi with a fix for this issue

thijstriemstra, 2018-09-26:
@viktor6 use this:

```
git clone https://github.com/rm-hull/luma.led_matrix.git
cd luma.led_matrix
sudo pip install -e .
```

max7219 is the old name for this library.

viktor6, 2018-09-27:
Give instructions for the current version of max7219