rm-hull/luma.led_matrix #166: Text scrolling throught 4 8x8 (MAX7219)

GYJohn, 2018-11-22:
#### Type of Raspberry Pi
Got a RPi3.

#### Linux Kernel version
Linux raspberrypi 4.9.80

#### Actual behaviour & Expected behaviour
At the moment i execute  
`python matrix_demo.py --rotate 1`
the text is scrolling on all 8x8 separately and not on my 4x 8x8 MAX7219 as a "cluster".

I am rly new to the MAX7219 topic, so please be kind 🔢 

Thank you!

GYJohn, 2018-11-24:
Still need help :/

rm-hull, 2018-11-24:
If you run `python matrix_demo.py -h` it will show the options. If you have a 4x cluster, try adding `-n 4`.

Also, have a look at the documentation, daisy-chaining in particular: https://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#daisy-chaining

GYJohn, 2018-11-24:
@rm-hull Thanks for ur answer.
If i try: `python matrix_demo.py -n 4 --rotate 1` It scrolls on only one 8x8 :(

rm-hull, 2018-11-24:
What happens if you take off the `--rotate 1` flag?
How have you wired up your 4x "cluster" ?
Any pics of you board / wiring (you can drag/drop images to the github issues text field to upload them), or links to where you bought from would be useful

GYJohn, 2018-11-24:
I bought this one:
https://www.amazon.de/gp/product/B079HVW652/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

I can make a video later if this helps

rm-hull, 2018-11-24:
Looking at the video it is exactly the block orientation issue. 
Try running the demo with `python matrix_demo.py -n 4 --block-orientation=90` ... if that doesn't look right, try -90 instead

GYJohn, 2018-11-24:
You are right....... Damn it. 
Thank you!!

closed 2018-11-24 by GYJohn