rm-hull/luma.led_matrix #169: Possibility to use differents pins

SEMI-K, 2018-11-25:
Hi,
Thanks for this beautiful job !
Is it possible to connect the max7219 to others pins than the RPi pins 19, 24, 23 ? (for example 16, 20, 21)
Thanks for helping,
Jerome

rm-hull, 2018-11-26:
You can only use alternative SPI pins, see https://pinout.xyz/pinout/spi .. you would need to specify a different port and/or device though

SEMI-K, 2018-11-26:
OK, I understand ! Then, if I use the port 1, I've to use 
GPIO 20 instead of GPIO 10 (MOSI)
GPIO 16 instead of GPIO 8 (SPI CE0)
GPIO 21 instead of GPIO 11 (SPI CLK)
Let's try !
Thanks !

closed 2018-11-28 by rm-hull

paulvanderheijden90, 2019-10-31:
> OK, I understand ! Then, if I use the port 1, I've to use
> GPIO 20 (P38) instead of GPIO 10 (MOSI)
> GPIO 18 (P12) instead of GPIO 8 (SPI CE0)
> GPIO 21 (P40) instead of GPIO 11 (SPI CLK)
> 
> It's works great !
> Thanks !

Hello
So by changing   serial = spi(port=0, device=0, gpio=noop()). To 
  serial = spi(port=1, device=0, gpio=noop())
Will activate the second pins provide on te Raspberry.????

Is it also posible to use them both for the 7219 matrix?
So i can program 2 lines of matrixes. Like this:

`
# create matrix device
    serial1 = spi(port=0, device=0, gpio=noop())
    device1 = max7219(serial1, cascaded=n or 1, block_orientation=block_orientation,
                     rotate=rotate or 0, blocks_arranged_in_reverse_order=inreverse)
    print("Created device1")

    # create matrix device
    serial2 = spi(port=1, device=0, gpio=noop())
    device2 = max7219(serial2, cascaded=n or 1, block_orientation=block_orientation,
                     rotate=rotate or 0, blocks_arranged_in_reverse_order=inreverse)
    print("Created device2")

msg1 = "Slow scrolling: This will be matrix string no 1"
    print(msg1)
    show_message(device1, msg1, fill="white", font=proportional(LCD_FONT), scroll_delay=0.1)

msg2 = "Slow scrolling: this wil be matrix chain no2"
    print(msg2)
    show_message(device2, msg2, fill="white", font=proportional(LCD_FONT), scroll_delay=0.1)
`

Please let me know in this Will work?