rm-hull/luma.led_matrix #170: Implementing Code From DotMatrixTool

dh47, 2018-12-03:
Hello,

Is there anyway to take the code that has been generated from http://dotmatrixtool.com and implement it in python code for the dot matrix display?

rm-hull, 2018-12-04:
Absolutely!

This is pretty much how the legacy font definitions are created. See https://github.com/rm-hull/luma.core/blob/master/luma/core/legacy/font.py#L81 for example.

You could create your own design in the dotmatrixtool website as follows:

[image]

Which gives us:

```c
const uint_8 data[] =
{
0x00, 0x3e, 0x08, 0x3e, 0x00, 0x3e, 0x2a, 0x22, 0x00, 0x3e, 0x20, 0x20, 0x00, 0x3e, 0x0a, 0x0e
};
```

Then copy pasting into Python:

```python
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.led_matrix.device import max7219
from luma.core import legacy

MY_CUSTOM_BITMAP_FONT = [
    # Paste byte code in here, remember to use square brackets rather than curlies...
    [ 0x00, 0x3e, 0x08, 0x3e, 0x00, 0x3e, 0x2a, 0x22, 0x00, 0x3e, 0x20, 0x20, 0x00, 0x3e, 0x0a, 0x0e ]
]

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial)

with canvas(device) as draw:
    # Note that "\0" is the zero-th character in the font (i.e the only one)
    legacy.text(draw, (0, 0), "\0", fill="white", font=MY_CUSTOM_BITMAP_FONT)
```

This is it running in the emulator:

[image]

With the following command: `python3 examples/dotmatrixtool.py -f conf/led_matrix.conf --width=16 --height=8` (I will publish the example source code shortly, but it more-or-less looks like the python above)

sonic2wb, 2018-12-04:
totally Awesome @dh47 and I were working on this and you made it perfect!

sonic2wb, 2018-12-04:
https://cdn.discordapp.com/attachments/517528896451641359/519432829990469633/VID_20181204_033937996.mp4

There is a video of a small animation we made.

closed 2018-12-04 by rm-hull

Rishikeshdaoo, 2020-10-18:
This is an awesome feature!
I wanted to make sort of an animation where it iteratively goes through each frame of the dot matrix. (similar to how @sonic2wb has shown in the comment above, just a lot more frames, one after the other)

Any pointers on how that could be achieved?