rm-hull/luma.led_matrix #173: Snake layout with size other than 8*8

thomasw-mitutoyo-ctl, 2018-12-30:
#### Type of Raspberry Pi

Raspberry Pi 3.
Note that it's not a hardware issue. WS2812B works when size is set to 8*8

#### Linux Kernel version

4.14.79-v7

#### Background information

I have a 8*32 WS2812 board that is assembled in snake mode. You can find them on Ali Express by searching for "ws2812b 8x32".

#### Expected behaviour

I expected the LEDs to work with the setting

```
device=ws2812(width=32, height=8, mapping=UNICORN_HAT)
```

I use width=32 and height=8 as these are my physical dimensions and I use UNICORN_HAT because that should give me the snake layout.

#### Actual behaviour

There's an AssertionError in device.py in `__init__()` at

```
assert(self.cascaded == len(self.mapping))
```

followed by an AttributeError

```
'ws2812' object has no attribute '_ws'
```

As far as I understand, the object does not have the property, because the `__init__()` method did not run to the point where self._ws is initialized.

thomasw-mitutoyo-ctl, 2018-12-30:
Perhaps the fixed size array UNICORN_HAT could be replaced by a generator.

I also see that I can replace the `mapping` parameter by something I create myself.

So I had a look at the UNICORN_HAT array and I managed to create the same array by a generator and then increased the length of that generator. However, my first try behaved snakey. Therefore, I flipped every second 8 elements in the resulting array. I was really puzzled to see what happend. The result is even worse. It's not intuitive how the mapping works.

Until now I have tried 6 different generators but I was still unable to figure out the seuence needed for my LED panel.

rm-hull, 2018-12-30:
Have a look at https://github.com/rm-hull/luma.led_matrix/blob/master/examples/neopixel_crawl.py (change the cascaded=... to suit your purposes). This should give you an idea how your board is laid out

roboter-basteln, 2019-01-04:
@rm-hull: I followed that example and successfully created a mapping for `width=256, height=1`. All LEDs do light up in the order I want to. It worked exactly like expected, that is: how I see the wires on the PCB.

However, when I use the same mapping on `width=32, height=8`, the LEDs light up differently.

I have modified the code to do a 2D mapping as well, but as far as I understand, I can't put tuples into the mapping, just scalar values. I still didn't figure out the formula for mapping the x-y-coordinates to scalar values. I've tried the combinations `x*width+y`, `x*height+y`, `y*width+x` and `y*height+x`.

rm-hull, 2019-01-04:
Can you record and post a video of the neopixel crawl running on your 32x8 matrix please?

Also, can you share the mapping code you have tried please?

thijstriemstra, 2019-04-23:
ping @thomasw-mitutoyo-ctl

closed 2019-05-25 by rm-hull