rm-hull/luma.led_matrix #176: partial scroll

olober, 2019-01-18:
#### Type of Raspberry Pi
Pi Zero W / Linux 4.14.79+ 

Hello,
I use the lib with max7219 leds - 4 modules (32x8 pixels) without any problems. I use adapted code from the examples folder - I create virtual via viewport, use virtual.set_position - scroll is working perfectly well.
Now, I have an idea to use 8 modules (32x16 pixels) and I cannot make the partial scroll.
Is it possible to scroll only part of the display. eg. top part displays static text but bottom part has scrolling text. I know that theoretically I can connect the top/bottom part as two separate SPI devices, but this time I have another SPI device connected, and RPi has limit of just two. 
Of course it won't solve the problem with about eg. 12 modules (32x24 px).

Thank you.

olober, 2019-03-06:
I solved it pretty easily - there is no need to modify code - so I close the issue.

rm-hull, 2019-03-06:
Ok, glad you solved it. There is an example (https://github.com/rm-hull/luma.examples/blob/master/examples/image_composition.py) that someone contributed that does partial scroll.

How did you solve it? _(for the benefit of anyone else who comes looking to solve the same problem)_

olober, 2019-03-06:
Sample (not complete) code (you need import and initialize display - I used max7219):

`
from luma.core.sprite_system import framerate_regulator
from luma.core.render import canvas
from luma.core.legacy import textsize
from luma.core.legacy.font import LCD_FONT

######
text = "scrolling text"
regulator = framerate_regulator(30)
tx, ty = textsize(text, LCD_FONT)
dw = device.width
i = 0
while i <= tx + dw:
    with regulator:
        with canvas(device) as draw:
            text(draw, ( 1, 1), "STATIC", fill="white", font=LCD_FONT)
            text(draw, (dw-i, 9), text, fill="white", font=LCD_FONT)
        i += 1
`

pagaille, 2019-04-11:
> There is an example (https://github.com/rm-hull/luma.examples/blob/master/examples/image_composition.py) that someone contributed that does partial scroll.

Does it ? It was also my impression but as soon as ComposableImage is invoked in the code the display contents are completely cleared and only the scrolling text appears. The rectangle is redrawn aftermath.

Moko0815, 2021-08-16:
Please post your full code.
Thanks.