rm-hull/luma.led_matrix #177: support multiple WS2812 8x8 matrices

MrQwer, 2019-01-22:
Hi,

I've recently ordered 3 WS2812 8x8 matrices. I connect them through their DataOut and DataIn connectors. For WS2812 LEDs, there are two modes: **cascade** them or initialize the device like a Unicorn pHAT with **width** and **height**. 

The problem is: I can't daisy-chain several matrices.
I want all three matrices next to each other in a row, horizontally.
The best way is to initialize them with width and height.
In a horizontal orientation (width=16, height=8), only every second column gets lightened up.
With the vertical orientation (width=8, height=16), I can't rotate all matrices "at once", meaning not seperate but as a complete matrix.

I would be glad to see a function to cascade whole matrices with a particular layout (8x8), not only single LEDs.

#### Type of Raspberry Pi
Raspberry Pi Zero W

MrQwer, 2019-01-23:
I fixed it by creating a mapping like this:

	twoMatrices8x8 = [
		0,  1,  2,  3,  4,  5,  6,  7, 64, 65, 66, 67, 68, 69, 70, 71,
		8,  9,  10, 11, 12, 13, 14, 15, 72, 73, 74, 75, 76, 77, 78, 79,
		16, 17, 18, 19, 20, 21, 22, 23, 80, 81, 82, 83, 84, 85, 86, 87,
		24, 25, 26, 27, 28, 29, 30, 31, 88, 89, 90 ,91, 92, 93, 94, 95,
		32, 33, 34, 35, 36, 37, 38, 39, 96, 97, 98, 99, 100, 101, 102, 103,
		40, 41, 42, 43, 44, 45, 46, 47, 104, 105, 106, 107, 108, 109, 110, 111,
		48, 49, 50, 51, 52, 53, 54, 55, 112, 113, 114, 115, 116, 117, 118, 119,
		56, 57, 58, 59, 60, 61, 62, 63, 120, 121, 122, 123, 124, 125, 126, 127
	]

and created a device with 

device = neopixel(width=16, height=8, mapping=twoMatrices8x8)

closed 2019-01-23 by MrQwer

rm-hull, 2019-01-23:
That's exactly the way to do it. I was getting round to replying, but you beat me to it. Sorry