rm-hull/luma.led_matrix #180: How to create custom fonts for max7219 display and use it in luma library?

Chidhambararajan, 2019-03-18:
Hello there,

I am currently working on a project in which there is a requirement to show some symbols in the max7219 8x8 display which is connected to a rpi.

I am now currently ready with the hexadecimal codes for each symbol but I don't know how to implement the same through luma library. Is there a way to load such custom hexadecimal codes in luma library? Or do we have any other similar library (compatible with rpi) which provides features to load custom hexadecimal codes for the max7219 8x8 display??

Kindly reply ASAP. 

Thanks In Advance

rm-hull, 2019-03-18:
Have a look at the discussion in issue #170

Chidhambararajan, 2019-03-18:
Yeah thanks that helped a lot !!!

closed 2019-03-18 by rm-hull