rm-hull/luma.led_matrix #191: max7219 package on pypi

thijstriemstra, 2019-06-14:
when searching for luma.led_matrix, the max7219 package comes up first: https://pypi.org/project/max7219/

there's nothing on that pypi page (e.g. first paragraph) that tells me it's a deprecated package. Can this be updated and made more clear for new users?

rm-hull, 2019-06-14:
Alternatively we could just delete all versions on pypi (same goes for https://pypi.org/project/ssd1306/)

rm-hull, 2019-06-27:
@thijstriemstra Unless you (or anyone else) have specific objections, I will remove all max7219 and ssd1306 packages from pypi this weekend.

thijstriemstra, 2019-06-27:
sounds good. i don't think there's a pypi-redirect option like github has..

rm-hull, 2019-06-30:
Packages and all release versions removed from pypi.

closed 2019-06-30 by rm-hull

thijstriemstra, 2019-06-30:
RIP.