rm-hull/luma.led_matrix #199: Neopixel PWM pin change

bouchardh, 2019-08-11:
Hi,

The PWM pin is hard coded to 18 in the library and it is already covered by an old Displayotron 3000 on my raspberry pi.

Would it be possible to make this pin user selectable (12 in my case) as it's the only hardware PWM pin that is not covered.

Thank you

thijstriemstra, 2019-08-12:
> The PWM pin is hard coded to 18

confirmed: https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py#L227

rm-hull, 2019-08-12:
I think it can only be connected to certain pins.

In any regard, I'll have a look at making this configurable when I get back off holiday in a few weeks, unless any one else fancies submitting a PR in the meantime

bouchardh, 2019-08-13:
Agreed. From what I have found is there are only 4 pins thar can be used but saw conflicting information about pin selection.

All I can tell for now is that **18** (hard coded) and **12** (tested by myself by modifying your code) work. They were both in the lists I found and there should be 4 usable pins (Adafruit also lists **10** and **21**). 

I will have some time later this week to test it out.

bouchardh, 2019-08-23:
I was able to control the lights with pins 12, 18 and 21. I was able to start the code with pin 10 also but I only got all white out of it.

robi772, 2019-09-15:
That pins are individualy usable? Can use it for 2 x 1024 LED?