rm-hull/luma.led_matrix #206: Question: Is it possible to split the MAX7219 into individual LED lights, or connect individual LED lights to the MAX7219

FireWizard312, 2019-12-07:
So I recently wrote a code that would help identify where certain things are. The problem is, I use LED lights to tell where each of the items is. Since the MAX7219 can't be attached to each box that has each item, is it possible to separate the MAX7219 into individual LED lights that can be controlled using the same code for a regular MAX7219? Keep in mind that I need about 128 LED lights to make this work.

rm-hull, 2019-12-08:
If you can source a max7219 chip, and a bunch of LEDs then there is no reason why it wouldn't work. All the MAX7219 is multiplex over the 8 rows and 8 columns repeatedly. The LED matrix will be arranged something like this:

[image]

rm-hull, 2019-12-09:
To be fair, I would just buy a strip of ws2812b leds - they are wired in series so it would be much less hassle than lots of wires, and you would get programmable color RGBs to boot

FireWizard312, 2019-12-09:
Its just that i need 128 lights

closed 2020-10-18 by thijstriemstra