rm-hull/luma.led_matrix #208: Matrix is failing after some time.

paulvanderheijden90, 2019-12-16:
Hello.

My matrix is failing after some time. And wenn I restart the program it works fine for some time. 
and after some time it is failing again.

 
[image]

how can this happen ?

Here is the code.

> #!/usr/bin/python

# P2000 ONTVANGEN OP RASPBERRY PI 2 MET RTL-SDR
# https://nl.oneguyoneblog.com/2016/08/09/p2000-ontvangen-decoderen-raspberry-pi/
#
# vergeet niet deze regel verderop aan te passen aan je eigen RTL-SDR ontvanger (-p en -g):
# multimon_ng = subprocess.Popen("rtl_fm -f 169.65M -M fm -s 22050 -p 43 -g 30 | multimon-ng -a FLEX -t raw -",

import time
import datetime
import sys
import subprocess
import os
import re
import RPi.GPIO as GPIO
from datetime import timedelta
from dateutil import tz
from termcolor import colored
######################################
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.legacy import text, show_message
from luma.core.virtual import viewport
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT
from luma.led_matrix.device import max7219
######################################

# Define device settings
######################################
serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, width=80, height=16, block_orientation=-90)
virtual = viewport(device, width=80, height=16)
GPIO.setmode(GPIO.BCM)
GPIO.setup(17, GPIO.OUT)
GPIO.setup(27, GPIO.OUT)
GPIO.setup(22, GPIO.OUT)
GPIO.output(17,1)
GPIO.output(27,1)
GPIO.output(22,1)
######################################
line = "  "
mx_crit = True
mx_puls = True

capcode_avls_steenbergen= "1201999"#201431

def curtime():
    return time.strftime("%H:%M:%S %Y-%m-%d")
with canvas(virtual) as draw:
    text(draw, (4, 1), "Check Signaal", fill="white", font=proportional(LCD_FONT))
    text(draw, (42, 9), "00", fill="white", font=proportional(LCD_FONT))
    text(draw, (59, 9), "00", fill="white", font=proportional(LCD_FONT))
    text(draw, (4, 9), "Wait...", fill="white", font=proportional(LCD_FONT))
multimon_ng = subprocess.Popen("rtl_fm -f 169.65M -M fm -s 22050 -p 43 | multimon-ng -a FLEX -t raw -",
                               stdout=subprocess.PIPE,
                               stderr=open('error.txt','a'),
                               shell=True)

try:
    while True:
#        line = "FLEX: 2018-12-02  12:12:12 1600/2/A 11.059 [001201999] ALN P 2 Stormschade test Bredaseweg Akkermolenweg Zundert 203231 "
#        line = "FLEX: 2018-12-02  12:12:12 1600/2/A 11.059 [001201999] ALN P 2 Stormschade INCI-06 Bredaseweg Akkermolenweg Zundert 203231 "
        line = multimon_ng.stdout.readline()
        multimon_ng.poll()
        if line.__contains__("ALN"):
            with canvas(virtual) as draw:
                device.contrast(0 * 16)#tot 15
                text(draw, (40, 1), '20-1634', fill="white", font=proportional(LCD_FONT))
            if line.startswith('FLEX'):
                flex = line[0:5]
                timestamp = line[6:25]
                melding = line[59:]
                meldingp1 = line[63:]
                groupid = line[37:43]
                capcode = line[45:54]
                prio    = line[60:63]
                print(melding)
            time.sleep(3)

            if capcode_avls_steenbergen in (str(capcode)):
                GPIO.output(17,0)
                GPIO.output(27,0)
                print("17,27 aan")
                FMT="%d/%m/%y-%H:%M:%S"                                       #format waarin de tijd moet worden weergeven 
                old_tijd = timedelta(seconds = 0)
                rest_tijd = timedelta(seconds = 0)#defineer de rest tijd en zet deze op 0
                pinpuls_tijd = timedelta(seconds = 1)
                melding_tijd = timedelta(seconds = 600)            
                critical_tijd = timedelta(seconds = 330)
                start_tijd = datetime.datetime.now().strftime(FMT)            #bepaal het moment dat het bericht is binnen gekomen.
                pos=(melding.find("INCI-"))                                    #bepaal de positie van inci om dit door te geven op het aparte scherm
                if pos > 0:
                    inci= melding[pos+5:(pos+7)]                                        #bepaal inci
                    inci_BV= (int(inci)+20)                                            #bepaal inci Bevelvoerder
                else:
                    inci="  "
                    inci_BV= "  "
                lengte=len(melding)                                               #bepaal de lengte van het bericht
                while rest_tijd < melding_tijd:
                    for i in range (lengte*7):
                        act_tijd = datetime.datetime.now().strftime(FMT)
                        rest_tijd = (datetime.datetime.strptime(act_tijd,(FMT)) - datetime.datetime.strptime(start_tijd,(FMT)))
                        if rest_tijd > pinpuls_tijd:
                            if mx_puls:
                                GPIO.output(27,1)
                                mx_puls = False
                                print("27 uit")
                        with canvas(virtual) as draw:
                            device.contrast(15 * 16)#tot 15
                            t = str(rest_tijd)
                            if rest_tijd > critical_tijd:
                                if (rest_tijd!=old_tijd):
                                    if mx_crit:
                                        GPIO.output(22,0)
                                        mx_crit = False
                                        print("22 aan")
                                    text(draw, (-1, 8), chr(7), fill="white")
                                    text(draw, (33, 8), chr(7), fill="white")
                                    old_tijd = rest_tijd
                            text(draw, (7, 9), t[2:], fill="white", font=proportional(LCD_FONT))
                            text(draw, (42, 9), inci, fill="white", font=proportional(LCD_FONT))
                            #text(draw, (53, 0), chr(47), fill="white", font=proportional(SINCLAIR_FONT))
                            text(draw, (59, 9), str(inci_BV), fill="white", font=proportional(LCD_FONT))
                            if pos > 0:
                                text(draw, (80-i, 1), melding[:pos] + melding[pos+7:], fill="white", font=proportional(LCD_FONT))
                            else:
                                text(draw, (80-i, 1), melding, fill="white", font=proportional(LCD_FONT))
                        time.sleep(0.01)
                else:
                    device.clear()
                    mx_crit = True
                    mx_puls = True
                    GPIO.output(17,1)
                    GPIO.output(27,1)
                    GPIO.output(22,1)
                    print("17,27 en 22 uit")
                    print("FINSH: "+melding)
                    line="    "
        
        
except KeyboardInterrupt:
    GPIO.output(17,1)
    GPIO.output(27,1)
    GPIO.output(22,1)
    device.clear()
    GPIO.cleanup()

rm-hull, 2019-12-16:
I would put a few `print(...)` lines in the code, specifically before and after:

```python
        line = multimon_ng.stdout.readline()
        multimon_ng.poll()
```

as it could be the readline that is blocking

paulvanderheijden90, 2019-12-16:
Hello rm-hull,
Thank you for you're valued answer. 

The multimode code is correct. Like you can see I can comment the Multimon lines and just write a string in "line" wenn I do this the fail still occur 

I notice wenn the code is running ( line contain the values for the If state ment ) and I lower the value for 
'''
 with canvas(virtual) as draw:
                            device.contrast(15 * 16)#tot 15
'''

the problem is starting right a way.

paulvanderheijden90, 2019-12-18:
Hello.
I supply my print with a external 5v 5amp laptop supply. then I heave this problem. Wenn I use the raspberry USB to feed the power I down't have this problem. 

So the problem is there wenn I feed a external 5v. Is there a way to feed external. 
Ore do you heave a clue to fix this.

thijstriemstra, 2019-12-18:
> Wenn I use the raspberry USB to feed the power I down't have this problem.

If it works, why not use it?

paulvanderheijden90, 2019-12-18:
@thijstriemstra  
ofcourse this is the easy way. But I need some more devices to power.
so I need more amps

thijstriemstra, 2019-12-18:
Good to hear you got it to work.

closed 2019-12-18 by thijstriemstra