rm-hull/luma.led_matrix #209: Installing on Raspberry pi 4 warning: multi-line comment

gitalexa, 2019-12-18:
#### Type of Raspberry Pi
Raspberry Pi 4

#### Linux Kernel version
Linux raspberrypi 4.19.75-v7l+ #1270 SMP

#### Expected behaviour
After cloning it from git I run the setup with:
sudo python setup.py install

#### Actual behaviour
I getting some errors:

Running ws2812-1.0.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-I9ZERc/ws2812-1.0.0/egg-dist-tmp-Kewexm
warning: no files found matching 'ws2812-RPi.h'
warning: no files found matching 'ws2812-RPi.c'
warning: no files found matching 'LICENSE.txt'
In file included from lib/ws2812-RPi.c:1:
lib/ws2812-RPi.h:95:1: warning: multi-line comment [-Wcomment]
 // |   |   |  \  \___|  |_|  |  / /_/ \  ___/ \___ \
 ^
In file included from lib/ws2812-RPi.c:1:
lib/ws2812-RPi.h:123:1: warning: multi-line comment [-Wcomment]
 //  |    `   \  ___/|  |  |  |   |  \  ___/ \___ \  /  <_\ \/   \     /  / __ \|  | \/\___ \
 ^
lib/ws2812-RPi.h:389:1: warning: multi-line comment [-Wcomment]
 // |    |    \_   _____/\______ \    /   _____//  |_ __ ___/ ____\/ ____\
 ^
lib/ws2812-RPi.h:436:1: warning: multi-line comment [-Wcomment]
 //  |    |  \_/ __ \| __ \|  |  \/ ___\
 ^
lib/ws2812-RPi.h:456:1: warning: multi-line comment [-Wcomment]
 // |   |/    \|  \   __\ /    ~    \__  \\_  __ \/ __ |\ \/ \/ /\__  \\_  __ \_/ __ \
 ^
lib/ws2812-RPi.h:470:1: warning: multi-line comment [-Wcomment]
 //  |    |  / |  |_> > /_/ | / __ \|  | \  ___/  |    |___  |        \ |    `   \\___ \
 ^
lib/ws2812-RPi.h:481:1: warning: multi-line comment [-Wcomment]
 //  |        \|  |   |  | \  ___/\  \___|  |  \___ \
 ^
lib/ws2812-RPi.c: In function ‘reverseWord’:
lib/ws2812-RPi.c:49:16: warning: variable ‘bit’ set but not used [-Wunused-but-set-variable]
  unsigned char bit;
                ^~~
lib/ws2812-RPi.c: At top level:
lib/ws2812-RPi.c:147:1: warning: multi-line comment [-Wcomment]
 // |    |    \_   _____/\______ \    /   _____//  |_ __ ___/ ____\/ ____\
 ^
lib/ws2812-RPi.c:292:1: warning: multi-line comment [-Wcomment]
 //  |    |  \_/ __ \| __ \|  |  \/ ___\
 ^
lib/ws2812-RPi.c:471:1: warning: multi-line comment [-Wcomment]
 // |   |/    \|  \   __\ /    ~    \__  \\_  __ \/ __ |\ \/ \/ /\__  \\_  __ \_/ __ \
 ^
lib/ws2812-RPi.c: In function ‘initHardware’:
lib/ws2812-RPi.c:569:18: warning: suggest parentheses around comparison in operand of ‘&’ [-Wparentheses]
   if ((pfn >> 55)&0xfbf != 0x10c) {  // pagemap bits: https://www.kernel.org/doc/Documentation/vm/pagemap.txt
                  ^
lib/ws2812-RPi.c: At top level:
lib/ws2812-RPi.c:745:1: warning: multi-line comment [-Wcomment]
 //  |    |  / |  |_> > /_/ | / __ \|  | \  ___/  |    |___  |        \ |    `   \\___ \
 ^
lib/ws2812-RPi.c: In function ‘show’:
lib/ws2812-RPi.c:829:1: warning: "/*" within comment [-Wcomment]
 /**/
  
lib/ws2812-RPi.c:762:10: warning: unused variable ‘color’ [-Wunused-variable]
  Color_t color;
          ^~~~~
lib/ws2812-RPi.c:758:15: warning: unused variable ‘PWMWaveformBitPos’ [-Wunused-variable]
  unsigned int PWMWaveformBitPos = 0;
               ^~~~~~~~~~~~~~~~~
lib/ws2812-RPi.c:757:15: warning: unused variable ‘LEDBuffeWordPos’ [-Wunused-variable]
  unsigned int LEDBuffeWordPos = 0;
               ^~~~~~~~~~~~~~~
lib/ws2812-RPi.c: At top level:
lib/ws2812-RPi.c:839:1: warning: multi-line comment [-Wcomment]
 //  |        \|  |   |  | \  ___/\  \___|  |  \___ \
 ^
ws2812-RPi_wrap.c:125:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

thijstriemstra, 2019-12-18:
@gitalexa you skipped the installation instructions for dependencies, see https://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi

closed 2019-12-18 by thijstriemstra

gitalexa, 2019-12-18:
you are right... I followed this:
https://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi
and now it is working...

thankyou