rm-hull/luma.led_matrix #210: Seven Segment display problem

Daveolw, 2019-12-30:
Raspberry pi3 B+ with max7219/luma seven segment display controlled via SPI.

 I have 4 Digits with for buttons to get numbers 0-9999, pressing a button under each digit increases it by 1 and updates the display. It all works fine when I run the code,but once I stop the code and restart it some how it remembers what was on the display.
 I have "Device.clear" and set the digits to "0000", on restarting the code it shows "0000" but as soon as I press one button
 which should just change the digit from "0" to "1" it display what was there before I restarted the code, plus the "1". 
 ie if I had "3221" before restarting, if I press the 1st digit it goes to 4 where it should show 1 ("4221") should be "1000".
 Checking the variables (digit1_value and digit_total) used for the buttons and updating the display they are correct. 

 Thought it was the Max7219 because that does not power off when I stop the code so I put a switch in the positive  to turn it off, but that did not sovle it.
 If I shutdown the pi and turn off the power and start again it is ok until I once again stop and restart the code.

 code of buttons:-
def my_callbackp4(channel):   #1st digit song number
    global digit1_value
    digit1_value += 1
    if digit1_value  >= 10:
        digit1_value = 0
        
    seg_update()

code to update display:-
def seg_update():   #updates digits 1-4
    #print('seg_update')
    global digit1_value
    global digit2_value
    global digit3_value
    global digit4_value
    global digit_total
    digit_total = str(digit1_value) + str(digit2_value) + str(digit3_value) + str(digit4_value)     
    #print('digT = ', digit_total)
    seg.text[0:4] = digit_total

I do not know where it gets the previous data from, should I also switch the SPI connections to the max7219, is it a memory problem?
 I thought "Device.clear" cleared the memory.

 It is a bit inconvenient to have to keep shutting down and turning power off.

 Any ideas would be helpful,
 Thanks David

thijstriemstra, 2020-01-13:
can you paste a simple script that reproduces the problem? with this info it's not possible to figure out the problem.

Daveolw, 2020-01-13:
Thanks for the reply.  I have solved it, which was due to a second copy of programme running in the background.  I had the programme to start on boot up which I thought I then closed but it only closed the Tkinter part and not the Python code.

Thanks David

closed 2020-01-13 by Daveolw