rm-hull/luma.led_matrix #211: Does not work in daemon mode

pozsarzs, 2020-01-06:
RPi3, kernel v4.19.75, Python v3.7.3, display: 4x1088AS
Hi,
I want to use display with a program what run in daemon mode, but I have a problem. After changes daemon mode display freeze and program exiting. Before change works good.
I attached a test program in tar.gz package: [disptest.tar.gz](https://github.com/rm-hull/luma.led_matrix/files/4027663/disptest.tar.gz)
Thanks.

thijstriemstra, 2020-01-13:
can you paste a simple script here that demonstrates the issue?

pozsarzs, 2020-01-13:
Here is the test program:
```
#!/usr/bin/python3

import daemon
import time
import RPi.GPIO as GPIO
from luma.core.interface.serial import spi, noop
from luma.core.legacy.font import proportional, SINCLAIR_FONT
from luma.core.legacy import text
from luma.core.render import canvas
from luma.core.virtual import viewport
from luma.led_matrix.device import max7219
from time import gmtime, strftime

dir_log="/tmp/"
dbg_log="0"

# write steps to /tmp/debug.log and display
#dbg_log="1"

# write debug codes to display
def writedebugtodisplay(n):
  if dbg_log=="1":
    with canvas(virtual) as draw:
      text(draw, (0, 0), "D", fill="white", font=proportional(SINCLAIR_FONT))
      text(draw, (12, 0), '#'+n, fill="white", font=proportional(SINCLAIR_FONT))
  time.sleep(0.5)

# initializing display
def initdisplay():
  global virtual
  serial=spi(port=0, device=0, gpio=noop())
  device=max7219(serial, width=32, height=8, block_orientation=-90)
  device.contrast(5)
  virtual=viewport(device, width=32, height=8)

# write a line to debug logfile
def writetodebuglog(level,text):
  if level=="i":
    lv="INFO   "
    debugfile=dir_log+"debug.log"
    dt=(strftime("%Y-%m-%d %H:%M:%S", gmtime()))
    try:
      with open(debugfile, "a") as d:
        d.write(dt+'  '+lv+' '+text+'\n')
        d.close()
    except:
      print ("")

# main program
initdisplay()
writetodebuglog("i","00")
writedebugtodisplay("00")
with daemon.DaemonContext() as context:
  try:
    while True:
      writetodebuglog("i","01")
      writedebugtodisplay("01")
      time.sleep(1)
      writetodebuglog("i","02")
      writedebugtodisplay("02")
      time.sleep(1)
      writetodebuglog("i","03")
      writedebugtodisplay("03")
      time.sleep(1)
      writetodebuglog("i","04")
      writedebugtodisplay("04")
      time.sleep(1)
  except KeyboardInterrupt:
    GPIO.cleanup
exit(0)
```
See other files in attached tar.gz.

thijstriemstra, 2020-01-13:
`daemon.DaemonContext()` is from a different library that we haven't tested or support. Please ask that project or try stackoverflow.

rm-hull, 2020-01-13:
luma will register an atexit callback to shutdown when the program exits. There are several documented github issues that enumerate your options here should you not want to do that. See:
* https://github.com/rm-hull/luma.core/issues/74
* https://github.com/rm-hull/luma.oled/issues/140

I don't know anything about how the `daemon` library works, but if it somehow keeps the thread alive but manages to invoke the atexit callback, then this would account for the behaviour you are seeing.

Firstly try putting a `time.sleep(20)` before the exit line and see if you get the desired effect, up until the exit statement is executed. If it suddenly stops after 20 seconds we can be fairly certain this is the cause.

Then do as suggested here: https://github.com/rm-hull/luma.oled/issues/140#issuecomment-285804247 - basically have a do_nothing method that runs at clean-up.

pozsarzs, 2020-01-19:
Unfortunately it wasn't good, but I found a solution: If I make background service without daemon library then display is working.

I made a new example program package: It contents two different background service. First creates data, second displays it. First uses daemon library, second doesn't use. They are communicate via FIFO.

Thanks for help!

See attached file: [disptest2.tar.gz](https://github.com/rm-hull/luma.led_matrix/files/4083154/disptest2.tar.gz)

closed 2020-01-21 by pozsarzs