rm-hull/luma.led_matrix #215: Full Matrix module blanked.

paulvanderheijden90, 2020-02-03:
Hello,

i made a daisy chained 19 module led matrix.

I noticed that sometime 1 module is blanked out after some time.
so it is complete off.

I share the code. I use a program "multimon" to get information from a RTL Antenna.

[image]

This is the code.
``
import time
import datetime
import sys
import subprocess
import os
import re
import RPi.GPIO as GPIO
from datetime import timedelta
from dateutil import tz
######################################
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.legacy import text, show_message
from luma.core.virtual import viewport
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT
from luma.led_matrix.device import max7219
######################################

# Define device settings
######################################
serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, width=80, height=16, block_orientation=-90)
virtual = viewport(device, width=80, height=16)
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(17, GPIO.OUT)
GPIO.setup(27, GPIO.OUT)
GPIO.setup(22, GPIO.OUT)
GPIO.output(17,1)
GPIO.output(27,1)
GPIO.output(22,1)
######################################
mx_crit = True
mx_puls = True
mx_findcode=True
ii=0

capcode_avls_steenbergen= "1201999" #"1201585"

def curtime():
    return time.strftime("%H:%M:%S %Y-%m-%d")
#device.clear()
time.sleep(0.5)
with canvas(virtual) as draw:
    text(draw, (4, 1), "Check Signaal", fill="white", font=proportional(LCD_FONT))
#    text(draw, (2, 9), "00", fill="white", font=proportional(LCD_FONT))
 #   text(draw, (19, 9), "00", fill="white", font=proportional(LCD_FONT))
    text(draw, (32, 9), "Wacht...", fill="white", font=proportional(LCD_FONT))
multimon_ng = subprocess.Popen("rtl_fm -f 169.65M -M fm -s 22050 | multimon-ng -a FLEX -t raw -",
                               stdout=subprocess.PIPE,
                               stderr=open('error.txt','a'),
                              shell=True)
print("run")
try:
    while True:
#        line = "FLEX: 2018-12-02    12:12:12 1600/2/A 11.059 [1201999] ALN P 1 INCI-06 BR Gebouw Krommeweg Steenbergen NB 201634 "
        line = multimon_ng.stdout.readline()
        multimon_ng.poll()
        if line.__contains__("ALN"):
#            device.clear()
            time.sleep(0.5)
            with canvas(virtual) as draw:
                device.contrast(0 * 16)#tot 15
                text(draw, (24, 1), '20-1681*', fill="white", font=proportional(LCD_FONT))
            if line.startswith('FLEX'):
                flex = line[0:5]
                timestamp = line[6:25]
                melding = line[59:]
                meldingp1 = line[63:]
                groupid = line[37:43]
                capcode = line[45:54]
                prio    = line[60:63]

            if capcode_avls_steenbergen in (str(capcode)):
#                os.kill(multimon_ng.pid, 9)
                GPIO.output(17,0)
                GPIO.output(27,0)
                print("17,27 aan")
                FMT="%d/%m/%y-%H:%M:%S:%f"                                       #format waarin de tijd moet worden weergeven 
                old_tijd = timedelta(seconds = 0)
                beweegsnelheid = timedelta(seconds = 0)
                rest_tijd = timedelta(seconds = 0)#defineer de rest tijd en zet deze op 0
                pinpuls_tijd = timedelta(seconds = 1)
                melding_tijd = timedelta(seconds = 600)            
                critical_tijd = timedelta(seconds = 330)
                start_tijd = datetime.datetime.now().strftime(FMT)            #bepaal het moment dat het bericht is binnen gekomen.
                pos=(melding.find("INCI-"))                                    #bepaal de positie van inci om dit door te geven op het aparte scherm
                if pos > 0:
                    inci= melding[pos+5:(pos+7)]                                        #bepaal inci
                    inci_BV= (int(inci)+20)                                            #bepaal inci Bevelvoerder
                else:
                    inci="**"
                    inci_BV= "**"
                lengte=len(melding)                                               #bepaal de lengte van het bericht
                i=80
                while rest_tijd < melding_tijd:
                    act_tijd = datetime.datetime.now().strftime(FMT)
                    rest_tijd = (datetime.datetime.strptime(act_tijd,(FMT)) - datetime.datetime.strptime(start_tijd,(FMT)))
                    if rest_tijd > beweegsnelheid + (timedelta(microseconds =30000)):
                        if i < (-(round(lengte*5))):
                            i=80
                            
                        if rest_tijd > pinpuls_tijd:
                            if mx_puls:
                                GPIO.output(27,1)
                                mx_puls = False
                                print("27 uit")
                        if mx_findcode:
                            device.clear()
                            mx_findcode = False
                        with canvas(virtual) as draw:
                            device.contrast(15 * 16)#tot 15
                            dt = str(rest_tijd)
                            t = dt[:7]
                            if rest_tijd > critical_tijd:
                                if (rest_tijd!=old_tijd):
                                    if mx_crit:
                                        GPIO.output(22,0)
                                        mx_crit = False
                                        print("22 aan")
                                    if i % 2 == 0:
                                        ii=ii+1
                                        if ii>5:
                                            text(draw, (31, 8), chr(7), fill="white")
                                            text(draw, (65, 8), chr(7), fill="white")
                                            ii = 0
                                    old_tijd = rest_tijd
                            text(draw, (39, 9), t[2:], fill="white", font=proportional(LCD_FONT))
                            text(draw, (2, 9), inci, fill="white", font=proportional(LCD_FONT))
                            text(draw, (19, 9), str(inci_BV), fill="white", font=proportional(LCD_FONT))
                            if pos > 0:
                                text(draw, (i, 1), melding[:pos] + melding[pos+7:], fill="white", font=proportional(LCD_FONT))
                            else:
                                text(draw, (i, 1), melding, fill="white", font=proportional(LCD_FONT))
                        
                        i=i-1
                        beweegsnelheid=rest_tijd
                else:
                    device.clear()
                    os.kill(multimon_ng.pid, 9)
                    mx_crit = True
                    mx_puls = True
                    mx_findcode = True
                    GPIO.output(17,1)
                    GPIO.output(27,1)
                    GPIO.output(22,1)
                    print("17,27 en 22 uit")
                    print("FINSH: "+melding)
                    time.sleep(2)
                    os.system("shutdown -r now")
        
        
except KeyboardInterrupt:
    GPIO.output(17,1)
    GPIO.output(27,1)
    GPIO.output(22,1)
    device.clear()
    GPIO.cleanup()

`
it is always the same module. I try to change the module and wire's 

Can you check if I made some fails in the code ?

thijstriemstra, 2020-02-03:
I would make a much smaller and simpler test program and use that to find mistakes. Can you make a small example that reproduces the problem?

rm-hull, 2020-02-03:
I think it may be something to do with the state and interaction of variables `i`, `ii`, `rest_tijd`, `critical_tijd` and `old_tijd` in the while loop, rather than anything specific with the display driver

paulvanderheijden90, 2020-02-03:
@rm-hull  can you explain That

rm-hull, 2020-02-20:
I think you need to refactor your code into smaller functions/methods to make it more readable. 
There is a bug somewhere in your logic.

closed 2020-02-20 by rm-hull