rm-hull/luma.led_matrix #216: Horizontally flipped image

HappyWheels, 2020-02-20:
Hello, I have an issue with the image being horizontally flipped on my 5 module max7219 display.
I imagine the root cause is an incorrectly wired set of display modules, but for now my problem looks like it could be fixed in software, I have made some attempts but not been able to make it work (very new to python).

The 5 max7219 modules are connected to a **raspberry pi zero w, with kernel 4.19.93+, Python 3.7.3**

A basic bounding box showed the modules needed a +90 degree block_orientation which fixed the orientation.
```
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.led_matrix.device import max7219

serial = spi(port=1, device=0, gpio=noop())
#device = max7219(serial, cascaded=5) #first test
device = max7219(serial, cascaded=5, block_orientation=90)

from PIL import ImageFont

font = ImageFont.truetype("fonts/pixelmix.ttf", 8)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="black")
```
[image]
[image]

However a draw point at (0, 0) lights up the bottom left led
`
draw.point((0,0), fill="white")
`
[image]
And text displays horizontally flipped
`
draw.point((0,0), fill="white") 
draw.text((0, 0), "Hello", fill="white")
`
[image]

Again but with a legacy font

```
import time
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.led_matrix.device import max7219

serial = spi(port=1, device=0, gpio=noop())
device = max7219(serial, cascaded=5, block_orientation=90)

from luma.core.legacy import text
from luma.core.legacy.font import proportional, CP437_FONT, LCD_FONT

with canvas(device) as draw:
    text(draw, (0, 0), "Hello", fill="white", font=proportional(CP437_FONT))

time.sleep(4)
```

[image]

To try and fix the problem i have attempted to use a PIL.Image.transpose(Image.FLIP_TOP_BOTTOM) in a few ways in the preprocess function of device.py, eg
`
image = image.transpose(Image.FLIP_TOP_BOTTOM) 
return image
`
without success. 

Could I please get some assistance in fixing the flipped image problem in software? or pointed in the right direction? 
I am new to python, and i think the correct way to make changes in the device.py preprocess function would be through dependency injection using super from my code? trying changes directly in device.py was more of an attempt at a proof of concept. If you think using the PIL transpose method in the preprocess function is the right way of flipping the image, i would also appreciate knowing what you think is the most pythonic way of doing this without having to modify the luma module directly?

rm-hull, 2020-02-20:
Try `block_orientation=270`

Different breakout boards will be wired up (internally) differently, hence the need to have the block_orientation parameter in the first place!!

HappyWheels, 2020-02-20:
Thanks! I don't know why I only tried +-90, I looked at the preprocess function without ever taking in that it takes a variable as input. Shake my head. Thank you for making luma and answering the most tedious questions

thijstriemstra, 2020-02-20:
> Thank you for making luma and answering the most tedious questions

and thank you for taking the time to create such a detailed ticket :+1:

HappyWheels, 2020-02-21:
Sorry to reopen, but after adjusting device.py and altering to accept `block_orientation=270` (i realized it of course gives the same result as `block_orientation=-90`) a bounding block shows that a block_orientation=270 or -90 results in the modules needing the order to be reversed.
[image]
displaying text with `block_orientation=270`
[image]
so if i reverse the block order
`block_orientation=270, blocks_arranged_in_reverse_order=True` it will display
[image]
which like my first results requires a inverting of the entire image, this time it would need flipping left to right, where as the original `block_orientation=90` requires a horizontal flip for the text to display correctly. 
It is probably hard to see what's going on without the hardware to test, but i am happy to try more options and report back if you think there is another configuration worth trying. But it still looks like i will need the entire image to be flipped as well as block_orientation=90 to get the display to show correctly. 
The full code that gave the last image

```
import time
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.led_matrix.device import max7219

serial = spi(port=1, device=0, gpio=noop())
device = max7219(serial, cascaded=5, block_orientation=270, blocks_arranged_in_reverse_order=True)

from luma.core.legacy import text
from luma.core.legacy.font import proportional, CP437_FONT, LCD_FONT

with canvas(device) as draw:
    text(draw, (0, 0), "Hello", fill="white", font=proportional(CP437_FONT))

time.sleep(4)
```

Thanks again for taking the time to help me with this.

rm-hull, 2020-02-21:
Are they 5 separate modules that you have daisychained or 5 LED blocks on a single PCB?
Where did you buy them from (link) ?

HappyWheels, 2020-02-21:
They are in sets of 3 or 2 from here [max7219](https://www.electrodragon.com/product/dot-matrix-chain-display-kit-max7219-v2/) i joined a set of 3 to a set of 2 to make 5. I have some more, so i will find them and test them, or check i joined them up correctly, i did use another set of 5 i made up with a esp8266 running some Arduino code and they displayed ok (i think i did have to correct the orientation of the blocks, but can't remember if i needed to do anything else.)