rm-hull/luma.led_matrix #217: Led-matrix ramdomly on after switching power on

gitalexa, 2020-02-23:
this is not related to the project, but maybe someone has a solution...

i wrote some lines of python with luma.led_matrix to display  a clock. 
the program is working if i run it over terminal. but everytime if i switch on power of the raspberry, there are some of the matrixes switched on. sometimes all of them sometimes only a few. 

is there any possibility to get a "powered off matrix" state after power on raspberry?

if i do it with sudo reboot, then tis is not happening.

rm-hull, 2020-02-24:
When you power on the RPi, it will cause noisy spikes on the +5V power line - this may account for the glitches you are experiencing. Do you have any [decoupling capacitors](https://en.wikipedia.org/wiki/Decoupling_capacitor) installed across GND and VCC ?

thijstriemstra, 2020-10-18:
no feedback, closing.

closed 2020-10-18 by thijstriemstra