rm-hull/luma.led_matrix #218: OSError: [Errno 24] Too many open files

perrycannon, 2020-03-20:
#### Type of Raspberry Pi
Raspberry Pi Zero ver 1.3
Raspbian Stretch

#### Linux Kernel version
Linux raspberrypi4Port 4.19.102+ #1295 Thu Feb 6 15:38:35 GMT 2020 armv6l GNU/Linux

#### Expected behaviour
My program works, displays the date and time and a message. After 12 hours or more an exception is raised. I thought maybe the message was too long, shortened the message, same exception is raised.

#### Actual behaviour
My program works, displays the date and time and a message. After 12 hours or more an exception is raised. I thought maybe the message was too long, shortened the message, same exception is raised.

Traceback` (most recent call last):
  File "matrixTime.py", line 63, in <module>
  File "matrixTime.py", line 23, in demo
  File "/usr/local/lib/python3.5/dist-packages/luma/core/interface/serial.py", line 280, in __init__
OSError: [Errno 24] Too many open files

`#!/usr/bin/env python
# -*- coding: utf-8 -*-
# Copyright (c) 2017-18 Richard Hull and contributors
# See LICENSE.rst for details.

import re
import time
import argparse
import datetime
import pendulum

from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.virtual import viewport
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT

def demo(n, block_orientation, rotate, inreverse):
    # create matrix device
    serial = spi(port=0, device=0, gpio=noop())
    device = max7219(serial, cascaded=2 or 1, block_orientation=block_orientation,
                     rotate=rotate or 0, blocks_arranged_in_reverse_order=inreverse)
    #print("Created device")

    # start demo
    
    current_time = time.strftime("%H:%M:%S")
    current_date = time.strftime("%m/%d/%y")
    currentTime = pendulum.now()
    currentDay = currentTime.strftime("%A")
    #print(currentDay)
    msg = str(current_date) + " " + str(current_time) + " Today is " + str(currentDay)
    #print(msg)
    show_message(device, msg, fill="white", font=proportional(LCD_FONT), scroll_delay=0.1)
    time.sleep(2)
    #msg = "Today is " + str(currentDay)
    #msg = "O My Jesus, forgive us our sins, save us from the fire of hell, take all souls to Heaven, and help especially those most in need of Your mercy."
    #show_message(device, msg, fill="white", font=proportional(CP437_FONT), scroll_delay=0.05)
    #time.sleep(1)
    time_manila = pendulum.now('Asia/Manila')
    tm = time_manila.strftime("%H:%M:%S")
    dm = time_manila.strftime("%m/%d/%y")
    daym = time_manila.strftime("%A")
    msg = str(dm) + " " + str(tm) + " Today in Manila is " + str(daym)
    show_message(device, msg, fill="white", font=proportional(LCD_FONT), scroll_delay=0.1)
    time.sleep(2)
    
if __name__ == "__main__":
    parser = argparse.ArgumentParser(description='matrix_demo arguments',
        formatter_class=argparse.ArgumentDefaultsHelpFormatter)

    parser.add_argument('--cascaded', '-n', type=int, default=1, help='Number of cascaded MAX7219 LED matrices')
    parser.add_argument('--block-orientation', type=int, default=0, choices=[0, 90, -90], help='Corrects block orientation when wired vertically')
    parser.add_argument('--rotate', type=int, default=0, choices=[0, 1, 2, 3], help='Rotate display 0=0°, 1=90°, 2=180°, 3=270°')
    parser.add_argument('--reverse-order', type=bool, default=False, help='Set to true if blocks are in reverse order')

    args = parser.parse_args()

    while True:
        demo(args.cascaded, args.block_orientation, args.rotate, args.reverse_order)
#    except KeyboardInterrupt:
#        pass
        time.sleep(.1)

perrycannon, 2020-03-20:
Read the closed issue for the same problem. Corrected my program. Will reopen a new issue if occurs again.

closed 2020-03-20 by perrycannon

thijstriemstra, 2020-03-20:
> Read the closed issue for the same problem.

Which one is that?

perrycannon, 2020-03-20:
#190 OSError: [Errno 24] Too many open files
I moved this line `serial = spi(port=0, device=0, gpio=noop())' from the demo function to `if __name__ == "__main__":' The spi was created each time the demo function was executed from the `while True:` statement.