rm-hull/luma.led_matrix #220: MAX7219 fails with "error return without exception set"

DougieLawson, 2020-05-20:
#### Type of Raspberry Pi
 Raspberry Pi 3B

#### Linux Kernel version
 Linux endeavour 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux

Yes I'm running a 64-bit kernel and the very latest updated with `sudo rpi-update`. I'm quite happy to switch to running 5.4.42-v7 if you need that.

#### Expected behaviour
Your matrix demo program runs and does the usual stuff

#### Actual behaviour
I can get the same effect with python2 or python3 (I got the python3 version installed today with `sudo pip3 ...` to see if it made a different (it didn't)).

```
pi@endeavour:~/python$ ./matrix_demo.py
Traceback (most recent call last):
  File "./matrix_demo.py", line 120, in <module>
    demo(args.cascaded, args.block_orientation, args.rotate)
  File "./matrix_demo.py", line 20, in demo
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 281, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set
pi@endeavour:~/python$
```
I've hacked a very ugly C program to drive the matrix on the v8 5.4.42 kernel and that worked OK. You can have a copy of that if you need it (you won't).

rm-hull, 2020-05-22:
Works locally for me on Python3 on stock raspian. 

Luma.core has officially dropped support for Python2, and I am intending to drop it on all the sub-packages when I get the chance, so I haven't tried it - in fact, the latest version will refuse to install even on py27.

What versions of the luma software are you running. If you installed with pip, please try `pip list | grep luma` and paste the results here.

Have you googled for _"SystemError: error return without exception set"_ - it sounds like something fundamental went wrong inside the python runtime, and I suspect the fact you are using a 64-bit kernel might be playing a part.

Could you explain a bit more about your setup - i.e. how you built/installed the kernel ?

thijstriemstra, 2020-05-22:
> in fact, the latest version will refuse to install even on py27.

:( we should have updated and released all packages at once I guess.

DougieLawson, 2020-05-22:
It works for me on stock Raspbian (with 4.19.118+). I'm trying out the RPF 5.4.42 kernel (installed with `sudo rpi-update`) and trying to determine if that's something in your code or another kernel bug.  The kernel was built by the RPF/RPT folks from their normal git and published on https://github.com/hexxeh/rpi-firmware in the usual manner. I've been involved with testing the RPF early support kernels for the past eight years. See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769 for details of what's currently being tested.

```
root@endeavour:~ # pip list | grep luma
luma.core         1.12.0
luma.led-matrix   1.4.0
root@endeavour:~ # pip3 list | grep luma
luma              1.0.4
luma.core         1.14.0
luma.led-matrix   1.4.1
root@endeavour:~ #
```
I've stopped using python2 since Jan 1st (and I'm rewriting old code with python3 whenever I rediscover my old python2 stuff.

Googling for `SystemError: error return without exception set` tends to point to errors in C or CPython code. It's a wonderfully cryptic and arcane error (that mostly tells us nothing). Google really isn't giving up any secrets here.

BetaUtopia, 2020-06-18:
I had the same the error, however a kernel reinstall and reboot did the trick. Works like a charm.
`sudo apt-get install --reinstall raspberrypi-kernel`

```
  File "./matrix_demo.py", line 20, in demo
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 281, in __init__
    self._spi.cshigh = cs_high
```
It looks like the kernel when running latest `sudo rpi-update`.

DougieLawson, 2020-06-18:
It's still happening with the 5.4.47 kernel.

DougieLawson, 2020-07-20:
I think this one is fixed with the 5.4.51 kernel that the RPF/RPTL folks released today.

closed 2020-07-20 by DougieLawson