rm-hull/luma.led_matrix #221: Matrix on SPI1 vs SPI0

jwilts, 2020-05-28:
Hi this is a bit of a Noob Question but I am hoping that you can give me some direction on how to configure the matrix to run on SPI1 instead of SPI0

I have the same problem on both a Raspberry Pi3 and and PiZero W.

#### Type of Raspberry Pi
Pi 3 and Pi zero 

#### Linux Kernel version
I am running 4.18.97-v7+  #1294 as my kernel.

#### Expected behaviour

I am able to get the Max7219 8x8 matrix working properly, however I need to add another SPI device (RC422 RFID Reader) and the two do not seem to want to work sharing the CS pin (pin 24).
(Text on the Max7219 gets garbled and stilted)
From my research (and limited knowledge) I understand that I should be able to move to CS1 (PIN 26) and this in theory should solve my issue.

I am using this as my initialization code for the Max7219:
from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.virtual import viewport
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT

# create matrix device
serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=4, block_orientation=-90)

#### Actual behaviour
I have tried changing port=1, and device=1 but neither results in anything appearing on the matrix.

port=1 ==> "SPI Device Not Found"
device=1 ==> No error raised, but matrix does not update.

J0eK1ng, 2021-11-16:
Hi

I want to use my LED Matrix to display the temp of a MAX6675 Module with a Thermocouple. The MAX6675 will use SPI0 and I need to have the LED Matrix using SPI1. I've been trying to figure this out can you help?

Thanks

Kyuchumimo, 2024-05-30:
`luma.core.interface.serial` has a peripheral method by bitbang as SPI. This is the only way I see that can solve this problem without the help of the maintainers or contributors who seem to be either no longer active or have abandoned the library.

Instead of defining a serial object as you would normally do:
```
from luma.core.interface.serial import spi
...
'serial = spi(port=0, device=0, gpio_DC=23, gpio_RST=24)', 
```

Now you do it like this:
```
from luma.core.interface.serial import bitbang
...
'serial = bitbang(SCLK=21, SDA=20, CE=16, DC=23, RST=24)',
```

Please note that the bitbanging method is slower, as the interface is via software, but the advantage is that you can use any of your unused GPIO pins.

If you need more information, check the docsting of the bitbang class:
https://github.com/rm-hull/luma.core/blob/master/luma/core/interface/serial.py#L148

Kyuchumimo, 2024-05-30:
Ok, I found a way to use SPI1.

First check if you have SPI1 enabled. This is done in the terminal with the command `ls -l /dev/spi*`.

All SPI ports and devices should be displayed.
If SPI1 is enabled, you should see at least one that says /dev/spidev1.*.

If not, it is necessary to activate it. In the terminal type `sudo nano /boot/config.txt`. To the end of the text document add `dtoverlay=spi1-1cs`, then press ctrl+s to save and ctrl+x to exit. For the changes to take effect, it is necessary to reboot with `sudo reboot`.

The default SPI1 pins are as follows:
- MOSI - pin 38
- MISO - pin 35
- CE0 - pin 12
- CLK - pin 40

If you wish, you can also change the pin of CS0 by adding `,cs0_pin=<pin#>` to the line

now check again if SPI1 is already there with `ls -l /dev/spi*`.
you should see at least one line ending in /dev/spi1.<d>.
see that the format is spi<p>.<d>.

in the spi object, in the port parameter you will put <p> (in this case 1 of SPI1) and in device you will put <d> (as in this example we only enable 1 CS, the value is 0).
```
from luma.core.interface.serial import spi
...
'serial = spi(port=1, device=0, gpio_DC=23, gpio_RST=24)', 
```

References:
https://github.com/rm-hull/luma.examples/issues/78
https://raspberrypi.stackexchange.com/questions/73346/how-to-enable-spi1-and-spi0-at-the-same-time#73844
https://blog.stabel.family/raspberry-pi-4-multiple-spis-and-the-device-tree/