rm-hull/luma.led_matrix #225: luma.led_matrix not working on ZeroSeg 7-segment device after Raspberry Pi kernel upgrade

linuxminty, 2020-07-22:
#### Type of Raspberry Pi
1B+ (first generation B+)

#### Linux Kernel version
`Linux raspberrypi-B-Plus 5.4.51+ #1326 Fri Jul 17 10:39:45 BST 2020 armv6l GNU/Linux
`
#### Expected behaviour
I use the ZeroSeg to display time and date with code from luma.led_matrix examples. All worked fine with no problems until the recent installation of kernel 5.4.51.

#### Actual behaviour
Now when the code is run, I get the following:
```
Traceback (most recent call last):
  File "luma7datetime.py", line 58, in <module>
    main()
  File "luma7datetime.py", line 47, in main
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 285, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set
```
To try to solve, I've reinstalled luma.led_matrix; ran `rpi-update` to get latest firmware; used `sudo pip3 install --upgrade --force-reinstall luma.led-matrix luma.core` and `sudo pip3 install --upgrade --force-reinstall luma`. 

I see two closed issues with the same problem, but luma.led_matrix is not working on my Pi1B+. Help would be greatly appreciated.

linuxminty, 2020-07-25:
Updated/upgraded to kernel `5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux
`. Trying to run luma.led_matrix still results in system error:
```
Traceback (most recent call last):
  File "luma7datetime.py", line 58, in <module>
    main()
  File "luma7datetime.py", line 47, in main
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 285, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set

```

HelplessIdiot, 2020-07-25:
I have the same problem as you and did all you tried to fix it, plus reinstalling the rpi-kernel. Still, no improvement. So yes, this problem is still actual and affetcs (at least) two users, most likely more.

rm-hull, 2020-07-25:
The problem is actually occurring in a library called spidev. The exact problem looks to be some low level issue with C interfacing that Spidev does with the kernel.

What version of spidev do you have installed. Try: `pip3 list | grep spidev` to see the version.

If it shows 3.5, perhaps downgrade to 3.4 and retest ... this was released fairly recently with a change log of “fix memory leaks”. It is possible that there is some incompatibility in spidev and the version of the kernel you are using.

linuxminty, 2020-07-25:
Thank you for your reply. I do have spidev 3.5 installed:
```
~ $ pip3 list | grep spidev
spidev            3.5
```
However, I don't know how to downgrade to version 3.4. I need specific guidance on how to do that. Thank you in advance for your patience and assistance in getting this sorted.

rm-hull, 2020-07-25:
Try `pip3 install spidev==3.4.0 --force-reinstall`

linuxminty, 2020-07-25:
Hooray! That seems to have worked! At least it didn't return a Traceback error immediately as was the case previously. 

Thank you again for taking the time to respond, troubleshoot and solve this. I'm sure others who had the same problem will be grateful too for your excellent support. It's good to have the time/date display working once again on the ZeroSeg running Python3. Many, many thanks!

rm-hull, 2020-07-26:
It might be worth opening an issue on the spidev github issues page. In the meantime I will fix the spidev dependency version in luma.core to be <=3.4

Gadgetoid, 2020-07-27:
It is my understanding that the use of `cshigh` is no longer supported in any meaningful way as of kernel 5.4 and should be removed from this library- see: https://github.com/raspberrypi/linux/issues/3745

Switching to py_spidev <= 3.4 doesn't fix the problem so much as sweep the error under the rug. If you attempt to use any device that *requires* an active-high chip-select then it will fail.

The most correct behaviour is probably to remove any opinion about chip-select polarity from this Luma altogether- it should be set via device-tree or fixed by using inverting logic in hardware implementations.

rm-hull, 2020-07-27:
Understood. I shall remove the use of `cshigh` from this codebase.

But meanwhile, getting and setting `cshigh` is still part of the public interface on spidev, so while I can quash this unsupported behaviour within luma libraries, other folks using spidev elsewhere may undoubtedly hit the same problem... surely it should be removed from spidev if it is no longer supported by the underlying ioctls in the kernel?

Gadgetoid, 2020-07-27:
You're not wrong- I'm only just getting to grips with the issue and obviously have to be cautious about suggesting the deprecation of a feature that still works on older systems (afaik) and that is still used in existing libraries. There will definitely be a corresponding change in py_spidev, I just don't know what shape that will take yet.

Not entirely happy about the rug being pulled from under our feet like this, but I suppose it could always be argued (aha if only I had time) that I should be more up-to-date with core Linux changes. Times haven't exactly been the most forgiving!

Anyway don't take my word as gospel here, I've also been blind-sided by this chage and am still figuring out how best to deal with it.

thijstriemstra, 2020-07-27:
> Understood. I shall remove the use of cshigh from this codebase.

What about checking for the error instead?

rm-hull, 2020-07-27:
Note to self: SPI CSI was introduced in https://github.com/rm-hull/luma.core/pull/118

Gadgetoid, 2020-07-27:
Looks like this old, closed issue is also related: https://github.com/rm-hull/luma.led_matrix/issues/220

To boil the problem down to its simplest possible case, here's how mine behaves on Kernel 5.4.51 as mentioned by @DougieLawson -

```
pi@raspberrypizero:~ $ uname -a
Linux raspberrypizero 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
pi@raspberrypizero:~ $ python3
imPython 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import spidev
>>> dev = spidev.SpiDev(0, 0)
>>> dev
<SpiDev object at 0xb66b6e48>
>>> dev.cshigh = True
>>> dev.cshigh = False
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SystemError: error return without exception set
>>>
```

Since the default value for `cs_high` is `False`, I would expect most initialisations of the `serial` interface to fail with this error.

@DougieLawson is it fixed for you, or are you using `True` in this case and narrowly skirting the error?

DougieLawson, 2020-07-27:
Python3 works when I revert spidev to v3.4. Python2 doesn't work whatever I do (but I don't think I care about python2 enough to look any deeper).  I think someone needs to open an issue with the spidev folks as it looks to me like their code is broken on the 5.4.51 kernel. 

SPI works OK when I drive it with a C/C++ program.

thijstriemstra, 2020-07-27:
Also see https://github.com/doceme/py-spidev/issues/106

AdriaanKuipers, 2020-09-10:
I can't get past this hurdle:

`Linux zero 5.4.61+ #1339 Tue Sep 1 18:39:45 BST 2020 armv6l`

```
Traceback (most recent call last):
  File "max.py", line 7, in <module>
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 289, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set
```

spidev is on 3.4 `pip3 install spidev==3.4 --force-reinstall`
Did the same for luma.led-matrix, luma.core and luma

Has anyone got this running on a zero?

xMadNet, 2020-09-16:
Linux XPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
spidev 3.4
Raspi 3B+

But still also get same error
Traceback (most recent call last):
  File "sevensegment_demo.py", line 117, in <module>
    main()
  File "sevensegment_demo.py", line 71, in main
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python3.7/dist-packages/luma.core-1.16.1-py3.7.egg/luma/core/interface/serial.py", line 289, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set

infinic, 2020-10-10:
The same problem after upgrade packages with OLED 1306.

```
Traceback (most recent call last):
  File "./oled.py", line 14, in <module>
    serial1 = spi(port=0, device=0, cs_high=False, gpio_DC=24, gpio_RST=25)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 285, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set
```

closed 2020-10-11 by rm-hull