rm-hull/luma.led_matrix #226: Need assistance as to why this is not working now

bingbongosu, 2020-07-27:
Need help identfiy what went wrong.  I followed the instructions step by step.  Looked at previous issues and downgraded pip3 to 3.4

#### Type of Raspberry Pi: 3B
#### Linux Kernel version: Linux raspberrypi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux
#### Expected behaviour: light up of 8x8 matrix
#### Actual behaviour: this is what i get.
pi@raspberrypi:~/max7219 $ python examples/matrix_demo.py
Traceback (most recent call last):
  File "examples/matrix_demo.py", line 122, in <module>
    demo(args.cascaded, args.block_orientation, args.rotate, args.reverse_order)
  File "examples/matrix_demo.py", line 20, in demo
    serial = spi(port=0, device=0, gpio=noop())
  File "/usr/local/lib/python2.7/dist-packages/luma.core-1.14.1-py2.7.egg/luma/core/interface/serial.py", line 285, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set

thijstriemstra, 2020-07-27:
Duplicate of #225 

Try this @bingbongosu: https://github.com/rm-hull/luma.led_matrix/issues/225#issuecomment-663915929

closed 2020-07-27 by thijstriemstra

bingbongosu, 2020-07-27:
Please forgive my ignorance, why was this closed and re-referenced to 225.  I tried the solution in 225 it does not seem to fix the issue.

bingbongosu, 2020-07-27:
maybe my issue is from something else? I also get this, but I thought it was related to 225 as well

pi@raspberrypi:~/max7219 $ sudo -H pip install --upgrade luma.led_matrix
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: luma.led_matrix in /usr/local/lib/python2.7/dist-packages/luma.led_matrix-1.5.0-py2.7.egg (1.5.0)
ERROR: Package 'luma.led-matrix' requires a different Python: 2.7.16 not in '>=3.5, <4'

thijstriemstra, 2020-07-27:
> Please forgive my ignorance, why was this closed and re-referenced to 225. I tried the solution in 225 

The command that I linked to:

```
pip3 install spidev==3.4 --force-reinstall
``