rm-hull/luma.led_matrix #228: CJMCU WS2812 supported now?

MarkEstrada, 2020-08-13:
Are the following RGB LED from ebay https://www.ebay.com/itm/1x-CJMCU-2816-WS2812B-16-Bit-RGB-Full-Color-LED-Driver-Module-Development-UA/283957716001?hash=item421d32e421:g:IPAAAOSwGY5fGp7t still not supported by this luma.led_matrix at this time?

I saw this old [Support for CJMCU 8x8 64 Bit WS2812 5050 RGB LED Driver Development](https://github.com/rm-hull/luma.led_matrix/issues/95) and it seems to be closed.

Wondering if there has been any development at this time?
I happen to have bought some of this and I am finding out a way on how to power this with my RPI Zero W?

If not do you have an idea how can I drive this with my Pi?

Thanks

rm-hull, 2020-08-13:
It should work, But you would need to drive it with a decent power supply and level shift the PWM signal to 5v.

MarkEstrada, 2020-08-13:
> It should work, But you would need to drive it with a decent power supply and level shift the PWM signal to 5v.

Wow!  You have an awesome code in here by the way.  Thanks for maintaining this library.

A couple of questions though...
To drive this I can use the following sample code right? 
[neopixel_demo.py](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/neopixel_demo.py)

I haven't thought about level shifting the PWM signal of GPIO18..I have seen some links where level shifting was not applied but just would like to know what is the reason for this?

These rgb leds only has 3 pins DIN, +5V and GND, I supposed they mapped to the neosegments [https://luma-led-matrix.readthedocs.io/en/latest/install.html](https://luma-led-matrix.readthedocs.io/en/latest/install.html)?

Board Pin | Name | Remarks | RPi Pin | RPi Function
-- | -- | -- | -- | --
1 | GND | Ground | 6 | GND
2 | DI | Data In | 12 | GPIO 18 (PWM0)
3 | VCC | +5V Power | 2 | 5V0

Thank you and more power to this awesome library!

Gadgetoid, 2020-08-13:
@MarkEstrada level-shifting is - in theory - required to bring the logic voltage within the documented range of the WS2812. The lowest *documented* voltage for a logic high is VDD * 0.7 which at 5V is 3.5V. While the 3.3V output from the Pi falls somewhat short of this, in my experience WS2812 LEDs have always worked perfectly fine with 3.3V logic. This seems true of many things that require ~0.7 VDD logic levels. Your mileage may vary.

You should also ideally use an inline current-limiting resistor on the data line. This prevents inrush current causing overshoot and manifesting as signal issues. Something like 100Ohms should do.

MarkEstrada, 2020-08-14:
I am trying to run the  [neopixel_demo.py](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/neopixel_demo.py)
but I am encountering this problem

```
pi@raspberrypi:~/playground/neopixel $ python3 neopixel_demo.py
Traceback (most recent call last):
  File "neopixel_demo.py", line 19, in <module>
    device = neopixel(width=8, height=4)
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 223, in __init__
    ws = self._ws = dma_interface or self.__ws281x__()
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 264, in __ws281x__
    import _rpi_ws281x
ModuleNotFoundError: No module named '_rpi_ws281x'
Exception ignored in: <function ws2812.__del__ at 0xb5e075d0>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 322, in __del__
    if self._ws is not None:
AttributeError: 'ws2812' object has no attribute '_ws'
```
Not sure but I followed everything in [Installation section](https://luma-led-matrix.readthedocs.io/en/latest/install.html)

I have validated my luma library as the following

```
pi@raspberrypi:~/playground/neopixel $ pip3 freeze | grep luma
luma.core==1.15.0
luma.led-matrix==1.5.0
```

I have raspberry pi zero w running on headless setup

```
pi@raspberrypi:~/playground/neopixel $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/playground/neopixel $ cat /etc/debian_version
10.4
pi@raspberrypi:~/playground/neopixel $ uname -a
Linux raspberrypi 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux
```

Any hints?

MarkEstrada, 2020-08-14:
Sorry should have opened a new issue instead.  I am closing this question since my questions were answered already.  Thanks

closed 2020-08-14 by MarkEstrada