rm-hull/luma.led_matrix #229: Error running neopixel_demo.py

MarkEstrada, 2020-08-14:
I am trying to run the  [neopixel_demo.py](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/neopixel_demo.py)
but I am encountering this problem

```
pi@raspberrypi:~/playground/neopixel $ python3 neopixel_demo.py
Traceback (most recent call last):
  File "neopixel_demo.py", line 19, in <module>
    device = neopixel(width=8, height=4)
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 223, in __init__
    ws = self._ws = dma_interface or self.__ws281x__()
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 264, in __ws281x__
    import _rpi_ws281x
ModuleNotFoundError: No module named '_rpi_ws281x'
Exception ignored in: <function ws2812.__del__ at 0xb5e075d0>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 322, in __del__
    if self._ws is not None:
AttributeError: 'ws2812' object has no attribute '_ws'
```
Not sure but I followed everything in [Installation section](https://luma-led-matrix.readthedocs.io/en/latest/install.html)

I have validated my luma library as the following

```
pi@raspberrypi:~/playground/neopixel $ pip3 freeze | grep luma
luma.core==1.15.0
luma.led-matrix==1.5.0
```

I have raspberry pi zero w running on headless setup

```
pi@raspberrypi:~/playground/neopixel $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/playground/neopixel $ cat /etc/debian_version
10.4
pi@raspberrypi:~/playground/neopixel $ uname -a
Linux raspberrypi 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux
```

Any hints?

_Originally posted by @MarkEstrada in https://github.com/rm-hull/luma.led_matrix/issues/228#issuecomment-674004327_

rm-hull, 2020-08-14:
I think that the ws2812 and rpi_ws281x modules only gets installed on certain linux platforms: https://github.com/rm-hull/luma.led_matrix/blob/master/setup.py#L68-L71 shows:
```
extras_require={
        ':platform_machine=="armv7l" and platform_system=="Linux"': [
            'ws2812', 'rpi_ws281x'
        ],
```

And looking at the output of `uname -a` shows **arm6l**.
I dont recall exactly why we target arm7l specifically ... @thijstriemstra do you remember?
I cant see any issue with extending this for arm6l

MarkEstrada, 2020-08-14:
Additional comment also maybe this could help in debugging... I have tried the [matrix_demo.py](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py) and this works perfectly.
I am able to control everything using this awesome library of yours.

So I am quite sure that I followed the installation section as described.  Not sure what the problem is at the neopixel demo.
Thanks!

Gadgetoid, 2020-08-14:
Can confirm that all Pi Zero platforms support rpi_ws281x, and can't think of any good reason to limit to armv7l.

rm-hull, 2020-08-14:
@MarkEstrada the python setup script will only install the low level drivers for neopixels if the platform is **arm7l**, and your RPi Zero is showing the platform as **arm6l**, so it is not installing some critical components. I will get this fixed, but in the short term, you could just run the following:

```
sudo pip3 install ws2812 rpi_ws281x
```

Then try the demo script again - hopefully it should work.

MarkEstrada, 2020-08-14:
> @MarkEstrada the python setup script will only install the low level drivers for neopixels if the platform is **arm7l**, and your RPi Zero is showing the platform as **arm6l**, so it is not [...]

I did tried this but I am running into a different issue

```
pi@raspberrypi:~/playground/neopixel $ python3 neopixel_demo.py
Can't open /dev/mem: Permission denied
Traceback (most recent call last):
  File "neopixel_demo.py", line 19, in <module>
    device = neopixel(width=8, height=8)
  File "/home/pi/.local/lib/python3.7/site-packages/luma/led_matrix/device.py", line 258, in __init__
    raise RuntimeError('ws2811_init failed with code {0}'.format(resp))
RuntimeError: ws2811_init failed with code -5
Segmentation fault
pi@raspberrypi:~/playground/neopixel $ sudo python3 neopixel_demo.py
Traceback (most recent call last):
  File "neopixel_demo.py", line 13, in <module>
    from luma.led_matrix.device import neopixel
ModuleNotFoundError: No module named 'luma'
```

I have not been running my python scripts as sudo I think..Not sure if this is mandatory when using this library or I might be missing something?
I just started exploring Rpi development.

thijstriemstra, 2020-08-14:
> I dont recall exactly why we target arm7l specifically ... @thijstriemstra do you remember?

Not really.

> Can confirm that all Pi Zero platforms support rpi_ws281x, and can't think of any good reason to limit to armv7l.

Agreed. Pull request welcome.

MarkEstrada, 2020-08-15:
Got it to work!  Thanks for the help.  Awesome job guys!

rm-hull, 2020-08-15:
@MarkEstrada for the benefit of others who might be in the same situation ... what did you do to get it working?

MarkEstrada, 2020-08-15:
> @MarkEstrada for the benefit of others who might be in the same situation ... what did you do to get it working?

Oh my bad...sorry for not explaining what I did.

I followed your suggestion to install this
`sudo pip3 install ws2812 rpi_ws281x`

and run everything in sudo mode.  That did the trick!

Thank you for coming up with this awesome library!

closed 2020-08-15 by MarkEstrada