rm-hull/luma.led_matrix #232: What is the equivalent in the latest version of the library?

jwilts, 2020-09-29:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

Good day, quick question (Hopefully).  

In older versions of the LUMA library I was able to explicitly able to display a character on a specific position without disrupting everything else that was already on the cascaded LUMA Matrix.

import max7219.led as led
device = led.matrix(cascaded=4)
device.letter(0, ord('1'))
time.sleep(1)
device.letter(1, ord('2'))
time.sleep(1)
device.clear()

Unfortunately this code no longer works.  (No attribute Matrix, no attribute letter) etc.
I am looking for a very similar example to this using the latest version of the libraries.  I have read through about creating virtual viewports and that seems to be way more complex with a lot more overhead than what I need.   

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

I am looking to be able to post a specific character onto a 8x64 LED matrix in a specific spot without disrupting what is already on the matrix.  I need to put up "Place" information from a racetrack on the array as cars pass the finish line. With the old version of the library I could do this with code like this:

import max7219.led as led
device = led.matrix(cascaded=4)
device.letter(0, ord('1'))
time.sleep(1)
device.letter(1, ord('2'))
time.sleep(1)
device.clear()

But this code does not work with the newer versions of the library (at all).  No Attribute Matrix, No attribute Letter etc.   

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

I am able to get the matrix working using other samples so the matrix works and is wired properly, I am just struggling with the correct functions to call.

This code works.... (But I don't want to scroll)
from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.virtual import viewport
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT

import RPi.GPIO as GPIO
from mfrc522 import SimpleMFRC522

# create matrix device
serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=4, block_orientation=-90)

show_message(device, "test", fill="white", font=proportional(CP437_FONT), scroll_delay=0.1)

thijstriemstra, 2020-10-18:
> In older versions of the LUMA library

What version(s)?

> Unfortunately this code no longer works. 

Again, what version(s)?

rm-hull, 2020-10-19:
@jwilts i think you just want to use the legacy `text(...)` method instead, see documentation: https://luma-core.readthedocs.io/en/latest/legacy.html#luma.core.legacy.text