rm-hull/luma.led_matrix #234: led matrix (max7219) not working with raspberry pi 4

martinvoorbij, 2020-10-11:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

martinvoorbij, 2020-10-11:
I tried to get my max7219 working, after installing all the software none of  the examples is working.
I'm a beginner with the raspberry so it is hard to find out what is wrong. I think the problem is I have a raspberry pi 4.

this is the output from UNAME 

Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

I receive a system error : error return without exception after :

serial = spi(port=0 , device=0, gpio=noop()))

what is it what I did wrong ? 

Thnx

thijstriemstra, 2020-10-11:
@martinvoorbij Try installing latest master version of luma.core; `sudo -H pip3 install git+https://github.com/rm-hull/luma.core.git`

martinvoorbij, 2020-10-11:
@thijsriemstra, thanks for your help. Nothing changed, same error

thijstriemstra, 2020-10-11:
> I receive a system error : error return without exception after :

can you copy/paste the full error?

martinvoorbij, 2020-10-11:
i have no experience with copy/paste with raspberry but the error is like I

martinvoorbij, 2020-10-12:
goodmorning, below is a screen dump with the error I told you about. Hope you can help me to fix the problem but I guess it has something tot do with the version of my pi.

[image]

rm-hull, 2020-10-13:
@martinvoorbij did you manage to resolve the issue?

martinvoorbij, 2020-10-13:
@rm-hull, no I ddn't solve the problem. I think I have tried everything. I have not enough experience with python I think. I searched on the internet but I can't find people that got it working with my raspberry 4.

TheEclectic, 2020-11-18:
Is this related to issue  #225 ?

rm-hull, 2020-11-18:
It is/was the same issue; the issue has now been resolved with the latest luma versions, so assuming they are installed this should no longer occur. It is in no way specific to Raspberry Pi 4. The underlying cause was a library we use (spidev) became incompatible with a specific kernel version.

rursache, 2020-11-26:
i'm still having the same issue, i already updated my rpi zero packages and luma as well. maybe there are some old leftovers? any clue how to clean them up? thanks

martinvoorbij, 2020-11-26:
I tried everything, even with an other raspberry pi 4 but without succes.

closed 2022-01-09 by rm-hull