rm-hull/luma.led_matrix #235: Technically not an issue - looking for the old max7219 for Python 2.7

amdroid007, 2020-10-13:
According to the docs (luma.led_matrix/doc/install.rst) this library still supports Python 2.7 but pip refuses to install for python 3.5 and lower. Unfortunately I am adding a display to a robot that runs on python 2.7 code so I need to interface with python 2.7.

Looks like the old max7219 drivers are no longer available from pypi - is there a version of luma.led_matrix (and pre-requisities I guess) that can run on python 2.7? Or is the source for the old max7219 still available somewhere?

Thanks so much.

rm-hull, 2020-10-13:
It is possible that we missed updating the documentation or that read the docs did not properly regenerate the docs correctly, but the latest version does not support python 2.7. We took the decision to sunset that version along side python itself.

Version 1.4.1  of luma led matrix is the last version that supports py27.
You should be able to install as follows:
```console
$ sudo -H pip install luma.core==1.13.0 luma.led_matrix==1.4.1
```

Note that it wouldn’t be possible to upgrade any further than those versions.

Curious to know what hardware & OS your robot is using that prevents you from using python 3 ?

amdroid007, 2020-10-13:
Thank you! That (almost) worked. I keep getting an exception when running the demo code with the above set:
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 281, in __init__
    self._spi.cshigh = False
SystemError: error return without exception set

However, if I comment out that line (not sure how dangerous that is) the code seem to run ok. 

The robot that I am using is running on top of a Pi 3b. Python 3 is installed but the robot code that I am using is written in Python 2.7 and is over 3000 lines of code which would have taken a lot of effort to modify to Python 3, and it uses 2.7 libraries as well. Yes I realize 2.7 will be deprecated next January but your assistance with this is really appreciated!

A.

rm-hull, 2020-10-13:
There is a bug introduced with the latest kernel and the latest version of spidev (a library we rely on).

> However, if I comment out that line (not sure how dangerous that is) the code seem to run ok.

https://github.com/rm-hull/luma.core/pull/195 Does exactly what you just did, so I think you should be good.

Alternately, you should be able to downgrade spidev with: `sudo -H pip install spidev==3.4`

amdroid007, 2020-10-17:
Thank you! I am happy to report that the above process works beautifully with the robot so now I have successfully added a display that reduces the weight and power consumption without having to change any of the core code. All I had to do was to create a threaded version of the show methods so they did not block the robot operations (since the display needs to infinitely scroll in my application). Works beautifully.

Thank you!

closed 2020-10-17 by amdroid007