rm-hull/luma.led_matrix #237: IndexError: list index out of range

perrycannon, 2020-10-16:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
I am using a Raspberry Pi B Plus and a Raspberry Pi Zero. Same error on each.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

[matrixTimeCNN.zip](https://github.com/rm-hull/luma.led_matrix/files/5394082/matrixTimeCNN.zip)
This script gets the RSS feed from CNN. I cycle through to make one string of the 10 top headline titles. Save this as msg to print to the matrix led. The script has been running for 3 days with no issues. I have the same exception on the Raspberry Pi B Plus and Raspberry Pi Zero.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).
Traceback (most recent call last):
  File "matrixTime.py", line 75, in <module>
    demo(args.cascaded, args.block_orientation, args.rotate, args.reverse_order)
  File "matrixTime.py", line 46, in demo
    show_message(device, msg, fill="white", font=proportional(CP437_FONT), scroll_delay=0.1)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/__init__.py", line 81, in show_message
    w, h = textsize(msg, font)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/__init__.py", line 29, in textsize
    src = [c for ascii_code in txt for c in font[ord(ascii_code)]]
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/__init__.py", line 29, in <listcomp>
    src = [c for ascii_code in txt for c in font[ord(ascii_code)]]
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/font.py", line 44, in __getitem__
    bitmap = self.font[ascii_code]
IndexError: list index out of range

The script will of course stop after the exception. When I restart the script on both Raspberry Pi's, the exception occurs at the first instance of show_message for the CNN RSS string. The prior show_message in my script scrolls the date and time which works, no exception is returned.

I do not understand why the exception occurs?

thijstriemstra, 2020-10-16:
the font you use probably doesn't contain all characters you're trying to display, hence the `IndexError`?

rm-hull, 2020-10-18:
@thijstriemstra we could probably throw a more user-friendly error message that indicates the character that is not supported in the font

thijstriemstra, 2020-10-18:
Sounds good.

rm-hull, 2020-10-18:
@perrycannon i had a look at your script, and it is likely that there are unicode characters that are not supported by the bitmap font. There is a font wrapper class called `tolerant` which will substitute a replacement character if the relevant one is not found. Use as follows:

```python
from luma.core.legacy.font import proportional, tolerant, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT

...

show_message(device, msg, fill="white", font=proportional(tolerant(CP437_FONT)), scroll_delay=0.1)
```

See the docs for more information: https://luma-core.readthedocs.io/en/latest/legacy.html#luma.core.legacy.font.tolerant

perrycannon, 2020-10-18:
My next step was to try and iterate through to find which character was causing the exception.
Thank you for taking the time to help with my issue. I really do appreciate your help.
Perry

closed 2020-10-19 by rm-hull