rm-hull/luma.led_matrix #239: Cascading issue

TonyHSmith, 2020-10-25:
Let me start by saying a big "Thank you" to Richard Hull and all contributors for the excellent LUMA toolset. It really does make life so much easier when working with seven-segment and matrix displays. 

I am running LUMA (V1.5.0) on a Pi 3B+. (Linux raspberrypi 5.4.51-v7+ #1333). I have a question concerning the cascading of devices.

I have two displays (each consisting of a MAX7219 and an eight-digit, seven segment display) connected in a simple daisy chain. Both chips share the same CLK and LOAD signals. The MAX7219 in the first display takes its data directly from the Pi's MOSI output. The MAX7219 in the second display takes its data from the DOUT signal on the first MAX7219. 

When I run the sample program shown in section 3.2 ("7-Segment LED Displays") on page 13 of the "Luma.LED Matrix Documentation" (release 1.5.0) with seg.text = "Hello world" exactly as shown in the example, I get:

rld***** Hello*wo 

(where * indicates a blank character). 

In other words, the first part of the string is displayed on the second display in the chain, and the second part of the string is displayed on the first display. If I change the string to:

seg.text = "ABCDEFGH12345678"

I get: 12345678 on the first display, and: ABCDEFGH on the second.

Is this the way the module is supposed to work? It seems counterintuitive. I would have expected the first group of eight characters in the string to be displayed by the first MAX7219 in the chain, and the second group of characters to be displayed by the second MAX7219. 

I have been unable to determine what would happen with three or more displays connected in a cascade because, er, I don't have three or more displays, but I expect I would see something similar.

Thanks again for a superb piece of software.

Tony Smith.

rm-hull, 2020-10-26:
There are some differences in how some breakout boards have their LEDs 'oriented' with regards the MAX7219 driver chips, and there are plenty of folks reporting this if you look at past issues. As a result we added an extra capability called `block_orientation` (see documentation here: https://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#trouble-shooting-common-problems)

However, more or less all of the existing issues relate to the 8x8 LEDs rather than the 7-segment, so your case may be a novel issue.

I would expect that the display that shows `Hello*wo` is the one that is connected to the RPi breakout pins, and the one that shows `rld*****` is daisy chained off of the first display. Can you confirm this? Adding a clear photo of your wiring to this issue would be useful here,

Could you: 
* let me know a link/URL where you bought the 7-seg breakout board 
* try using block_orientation with values of 0, -90, 90 and 180 to see if that solves it for you

TonyHSmith, 2020-10-26:
Hi Richard,

Many thanks for your prompt response - much appreciated.

I understand the problems that can be encountered when using LUMA with 8x8 matrix modules. I have used them myself and found that the orientation problem can be fixed so easily using the 'block_orientation' feature (very neat and extremely useful).

However, to be clear, my issues here concern only eight-character, seven-segment displays similar to the type shown on page 13 of the "Luma.LED Matrix Documentation".

I have attached an annotated photo of my breadboard which consists of two 'displays'. At the top left of the breadboard is a 74HCT14 device that is used simply for buffering and for level shifting the Pi's 3.3V signals to the 5V required by the MAX7219 which is located immediately to the right of the HCT chip. This MAX7219 drives the larger 'first display' in the centre of the breadboard. 

The DOUT signal from the MAX7219 feeds the DIN input on the second display which consists of the module located at the bottom of the breadboard. The breadboard is a bit 'messy', so for clarity I have attached a simple block diagram which outlines the basic interconnections between the displays and the Pi.

As you can (hopefully) see from the photo, 'rld*****' is displayed on the first display, and 'Hello*wo' on the second. I should mention that I have tested each of the displays individually and they work perfectly with no problems at all.

I've also attached a screenshot of the program used to test the displays (run from Thonny on my Pi 3B+).

Please let me know if you require any further information.

Best regards,

Tony.

[image]
[image]
[image]

TonyHSmith, 2020-11-29:
Hi Richard,

Just a quick follow-up query to see whether you've been able to make any progress with this issue.

Have you been able to replicate my experiments and, if so, have you obtained similar results?

Kind regards,

Tony.