rm-hull/luma.led_matrix #241: matrix_demo dose not light up leds

uwe42, 2020-10-30:
Hello,

i am in the process of getting my project started, but hit a wall and could need some help.
I am running a raspberry pi zero with the latest debian server os and would like to controll a 32x8 matrix with 4 max7219.

What i have done so fare:

- activated spi via `sudo raspi-config`
- installed python3-dev, python3-pip
- updated pip `sudo -H pip3 install --upgrade --ignore-installed pip setuptools`
- installed lumz.led_matrix sudo -H pip3 install --upgrade luma.led_matrix`
- installed the matrix with the help of [pinout](https://pinout.xyz/pinout/spi#) and [luma.led_matrix/issues/160](https://github.com/rm-hull/luma.led_matrix/issues/160)
- cloned this repo

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

Raspberry Pi Zero wh

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 5.4.72+

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

if run `python3 matrix_demo.py --cascaded=4 --block-orientation=90`
the demo starts running, but the leds are all of (no python errors no stacktrace).

I am not sure which part is not working correct, python -> gpio -> spi -> leds and could need some help to figure this out.

thijstriemstra, 2020-10-30:
always start with double checking cabling/soldering. can you post some pictures?

uwe42, 2020-10-30:
[image]

TheEclectic, 2020-11-16:
This isn't wired up correctly, this is how it should be:
Pin-Name     RPI Pin    Function
1-VCC           2 or 4      5V0 power
2-GND          6              Gnd
3-DIN            19            GPIO 10 (MOSI)
4-CS              24            GPIO 8 (SPI CS0)
5-CLK            23            GPIO 11 (SPI CLK)

You have connected DIN and CS to the wrong pins.

closed 2021-07-18 by thijstriemstra