rm-hull/luma.led_matrix #245: Connecting 8 single seven segments to max7219

officialSeri, 2020-12-30:
Hello dear people

A couple of friends and i are currently working on connecting 23 single seven segments to a raspberry pi first we´re trying to connect 8 digits via one controller to the pi and make them light up to our wishes. 

We are following this guide : https://luma-led-matrix.readthedocs.io/en/latest/python-usage.html , but its made for an already connected 8 digit display which ours isnt (like ours is connected by hand (if this is understandable))

But we´re currently struggling on the following: firstly when we install spidev with :`sudo pip install spidev` we get the following error

```
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: spidev in /usr/lib/python2.7/dist-packages (3.4)
```

```
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.virtual import sevensegment
from luma.led_matrix.device import max7219

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=2)
seg = sevensegment(device)
```
this can run without an error tho  so all drivers of the max7219 should be installed (i guess)

Our goal final goal is to put numbers onto all 28 digits on our demands, so should be nothing "too fancy"

Thanks for your help already 

btw theres a picture of how it currently looks : https://ibb.co/yyv6M3F

thijstriemstra, 2020-12-30:
@officialSeri only Python 3 is supported.

officialSeri, 2020-12-30:
> @officialSeri only Python 3 is supported.

thanks for the quick response i think i managed to get this problem right by using pip3 install ...

how can i controll one digit segment tho? i searched for this but didnt find it on the tutorial

officialSeri, 2021-01-05:
> > @officialSeri only Python 3 is supported.
> 
> thanks for the quick response i think i managed to get this problem right by using pip3 install ...
> 
> maybe i messed smth up at the "connectinc seven segments to controller to raspberry pi" process
> 
> I would be really thankful if you could confirm that this code should light up all segments as 1s:
> 
> ```
> from luma.core.interface.serial import spi, noop
> from luma.core.render import canvas
> from luma.core.virtual import sevensegment
> from luma.led_matrix.device import max7219
> 
> serial = spi(port=0, device=0, gpio=noop())
> device = max7219(serial, cascaded=1)
> seg = sevensegment(device)
> 
> seg.text = "11111111"
> ```

Could someone please confirm or deny if my code is correct?

rm-hull, 2021-01-07:
Yes that code should work, but note that once the python script ends, it will shut down the display, so if you want it to persist once the program ends, you should set `device.persist = True` right after initializing.

I guess looking at your picture, all the LEDs light up when the program runs ... and you have wired up things manually, so i suspect you are going to have to do some hardware debugging.

Just buy a cheapo 7-seg breakout board from aliexpress or somewhere (e.g. https://www.aliexpress.com/item/4000813743744.html) - that will isolate any SPI and code issues first ... otherwise you won't know if the problem you have is with your wiring or something else

officialSeri, 2021-01-14:
Thanks for your response. We did a basic mistake and connected the seven segment to the controller the wrong way around (+ to - and - to +) After that we ran into hardware bugs... now we decided to buy a pretty similar seven segment to the one you sent.
can be closed

closed 2021-01-14 by officialSeri