rm-hull/luma.led_matrix #250: 7 Segment display writing backwards and needing leading spaces

matjspalding, 2021-04-04:
Hi

I've recently purchased a MAX7219 display and although it stated 
raspberry Pi Python libraries were avail they aren.t On the off chance I tried yours.  It is working after a fashion.  The display is writing backward (from right to left) and requires 3 leading spaces.

I have attached my code and a pic of the display.  I am using a Pi 4, Python 3.7.3.

As you can see the could should have displayed "HELLO" but I am getting "OLLE" on my display.  Am I doing something really stupid?

Thanks

Mat

[image]

import time

from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.virtual import viewport, sevensegment

def show_message_vp(device, msg, delay=0.1):
    # Implemented with virtual viewport
    width = device.width
    padding = " " * width
    msg = padding + msg + padding
    n = len(msg)

    virtual = viewport(device, width=n, height=8)
    sevensegment(virtual).text = msg
    for i in reversed(list(range(n - width))):
        virtual.set_position((i, 0))
        time.sleep(delay)

def show_message_alt(seg, msg, delay=0.1):
    # Does same as above but does string slicing itself
    width = seg.device.width
    padding = " " * width
    msg = padding + msg + padding

    for i in range(len(msg)):
        seg.text = msg[i:i + width]
        time.sleep(delay)
        
def main():
    # create seven segment device
    serial = spi(port=0, device=0, gpio=noop())
    device = max7219(serial, cascaded=1)
    seg = sevensegment(device)
    
    #print('Simple text...')
    #for _ in range(8):
    #    seg.text = "HELLO"
    #    time.sleep(0.6)
        
    print('Simple text...')
    for _ in range(8):
        seg.text = "   HELLO"
        time.sleep(0.6)
        
    
    #print('Scrolling alphabet text...')
    #show_message_vp(device, "HELLO EVERYONE!")
    #show_message_vp(device, "PI is 3.14159 ... ")
    #show_message_vp(device, "IP is 127.0.0.1 ... ")
    #show_message_alt(seg, "0123456789 abcdefghijklmnopqrstuvwxyz ABCDEFGHIJKLMNOPQRSTUVWXYZ")

if __name__ == '__main__':
    main()

rm-hull, 2021-04-05:
I guess the “OLLE” is the last 4 characters of “   HELLO” backwards. It is probably like this because of the way your device is wired up. You could try setting the block orientation flag on initilialization, see here for example: https://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#trouble-shooting-common-problems .. try setting it to 180 (or -180)

as to why you need the leading spaces ... max7219 IC’s control 8 seven segment devices serially, and I suppose the circuit you bought is wired up to the last 4 rather than the first four.

I would be curious to know where you got it from (I assume some sketchy eBay / Ali express seller or similar)

matjspalding, 2021-04-07:
Hi Richard

Thanks very much for your response.  I had tried that but it didn't 
work.  I was only messaging yourself on the off-chance you may have seen 
something similar in the past.  Psa the ebay link from where I purchased 
the display.  I'll be returning it and getting something else.

Thanks for your help

Mat

Hi! I found this on eBay and thought you might like it! Check it out 
now!  LED Bar Graph Power Display - RED - Arduino - ESP8266 - Raspberry 
Pi https://ebay.us/P1xc80

closed 2021-07-18 by thijstriemstra

pjaselin, 2022-01-03:
Hi @rm-hull I'm experiencing this as well using a 4 digit/7 segment display as well: https://www.mouser.com/ProductDetail/SparkFun/COM-11405?qs=sGAEpiMZZMvShe%252BZiYheisUaClgE6%2FvpGjlN%252BRV4q58%3D&countrycode=US&currencycode=USD

I'm having the additional issue that I can only show "."'s and 1's. Any other digit causes the display to die, any idea what might be causing that?

pjaselin, 2022-01-04:
Needed 4x the resistance! Still having the reversed text issue though

pjaselin, 2022-01-04:
And I just figured it out @matjspalding, if you reverse the digits, ie. switch D0 -> D7, D1 -> D6, etc, it'll fix this!

zanzeoo, 2023-03-29:
Same probleme here  plus i don't know where to modify thoses lines thx!