rm-hull/luma.led_matrix #251: The LED 8*8 Matrix look like malfunctioning

Leo-T1128, 2021-04-25:
Type of Raspberry Pi:  Raspberry Pi 4 Model B

Linux Kernel version : Linux raspberrypi 5.10.17-v7l+

Sample LED code:

from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.led_matrix.device import max7219
from luma.core import legacy

smile_FONT = [
[ 0x10, 0x26, 0x46, 0x40, 0x40, 0x46, 0x26, 0x10 ]
]

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=4, rotate= 0, block_orientation=-90)

with canvas(device) as draw:
legacy.text(draw, (1, 0), "\0", fill="white", font=smile_FONT)

When I trying that programme, the LED matrix just look like malfunctioning. It can't show the massage or words one section of all the LED just light up and I can't find the problem.

I already checks the SPI was enabled and all the pin connection was correct. Can anyone have a solution?

The situation when I trying the LED programme:
[https://youtu.be/9y75EUALCWA](url)

rm-hull, 2021-04-25:
Double check the pin connections. the program looks correct

Leo-T1128, 2021-04-26:
Thanks for the reply 

I already solve the problem, I found that after I unplug the additional fan (I bough from the Amazon and connect to the pin 4 and pin 6) The LED works fine. I think the error may cause by the EMI of the fan and disturb the signal.  

> Double check the pin connections. the program looks correct

closed 2021-04-26 by Leo-T1128