rm-hull/luma.led_matrix #252: Matrix max7219 : Rotate font for vertical display

fred2nice, 2021-07-10:
Hi Richard, 

Thank your for the module, everything works fine !

I put my 4x4 in vertical position, scrolling is ok but I would like to rotate letters to have  :
H
E
L
L
O

But I have this :
[image]

Do you have a magic option somewhere ? a special font ?

Kind Regards.
Frederic

thijstriemstra, 2021-07-11:
@fred2nice use the `rotate` parameter (in degrees):

```python
device = max7219(serial, width=w, height=h, rotate=180)
```

closed 2021-07-11 by thijstriemstra

fred2nice, 2021-07-11:
Hi, 

If I use rotate=2 ( 180°) without block_orientation=-90 the message is displayed on each block ( letters displayed correctly) , what I would like is a scrolling bottom to top, if I add block_orientation=-90 the scrolling is fine, but letters are in wrong orientation.
 
```
import time
from luma.core.interface.serial import spi, noop
from luma.led_matrix.device import max7219
from luma.core.legacy import show_message
from luma.core.legacy.font import proportional, CP437_FONT

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial,rotate=2,cascaded=4,block_orientation=-90)
show_message(device, "Hello !!!", fill="white", font=proportional(CP437_FONT), scroll_delay=0.05)
time.sleep(1)

```