rm-hull/luma.led_matrix #253: Is this library thread safe or thread compatible?

dachshund-digital, 2021-07-14:
Is this library thread safe or thread compatible?

rm-hull, 2021-07-18:
When interfacing with i2c or spi it is up to you to manage access to those resources. The core api and driver code aren’t really stateful so should be these safe. However, the diff-to-previous framebuffer is, so you would need to be careful there.

what application are you considering (or issue you are experiencing) that makes you ask this question?

dachshund-digital, 2021-07-24:
I noticed that the luma library fails to drive a display if the library is not initialized in the same thread context.  This was actually a code design issue in my code, initially, I was initializing the library in one context, but trying to display text via a different context, a sub thread.  This failed.  Even using proper shared memory or variables, no output on display.

The fix was simple, just making sure I initialize the library in the thread call back and use a queue to send display text strings to the thread call back, that goes into a controlled loop, that dequeues each message from the queue, thus, everything works.  So your suggestion appears true, as long as I have initialization done in the same thread that writes to the display, works.  I happen to be using SPI based, but i2c should be similar.

closed 2021-07-24 by dachshund-digital