rm-hull/luma.led_matrix #255: Segmentation fault with simple example

DanielSSilva, 2021-08-22:
Hi! Let me start by stating that I'm not super familiar with Python and I'm new to using this library. I'm using this library to develop a project with the NeoPixels. I've followed the installation guide and I'm able to get the expected results, however, everytime the program finishes I have a segmentation fault.
From what I was able to understand, this library uses the ws_281x C library, which can be where the problem relies....

#### Type of Raspberry Pi
Tested on a Rpi3 and Zero W

#### Linux Kernel version
Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux

#### Expected behaviour
Code run without any issue

#### Actual behaviour
Message is successfully displayed on the LEDs, but after that I get Segmentation fault, without any further information.

Here's the sample that I'm using

```python3
from luma.led_matrix.device import neopixel
from luma.core.legacy import show_message
from luma.core.legacy.font import proportional, SPECCY_FONT #https://luma-core.readthedocs.io/en/latest/_modules/luma/core/legacy/font.html

device = neopixel(width=16, height=7)
device.contrast(50)
def main():
    msg = "Hi!"
    show_message(device, msg, y_offset=-1, fill="green", font=proportional(SPECCY_FONT), scroll_delay=0.04)    

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        pass

```

thijstriemstra, 2021-08-22:
>  after that I get Segmentation fault, without any further information.

Can you paste at least the seg fault message? And what python 3 version are you using? can you try a different version?

DanielSSilva, 2021-08-22:
> Can you paste at least the seg fault message? 
That's the thing, the only message that I get is "Segmentation fault". There's no stack trace or anything related. 

>And what python 3 version are you using? can you try a different version?

I've just installed raspbian, so I'm assuming it's either 3.6 or 3.7. I'll try with another version when I get the chance. Should I try with some 3.x, or should it be with 2.x?

thijstriemstra, 2021-08-22:
Only 3.x is supported. Check version with `python --version` or `python3 --version`.

thijstriemstra, 2021-10-14:
> If you try latest Python 3.7.x I'm pretty sure the segfault will be gone.

No feedback, closing.