rm-hull/luma.led_matrix #259: AttributeError: module 'PIL.Image' has no attribute 'new'

unitythemaker, 2021-11-06:
#### Type of Raspberry Pi
Raspberry Pi Zero WH

#### Linux Kernel version
Linux raspberrypi 5.10.17+ #1414 Fri Apr 30 13:16:27 BST 2021 armv6l GNU/Linux

#### Expected behaviour
I'm expecting Python to run the code without errors. I inspected 10s of issues but didn't find any solution. Before getting that error I was getting something like: No module named luma.led.matrix found or same for luma.core.

#### Actual behaviour
```
pi@raspberrypi:~/luma.led_matrix $ sudo python3 -m pip install luma.core luma.led_matrix
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting luma.core
  Using cached https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
Collecting luma.led_matrix
  Using cached https://www.piwheels.org/simple/luma-led-matrix/luma.led_matrix-1.5.0-py2.py3-none-any.whl (19 kB)
Requirement already satisfied: cbor2 in /usr/local/lib/python3.7/dist-packages (from luma.core) (5.4.2)
Requirement already satisfied: pillow>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from luma.core) (8.4.0)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core) (3.4)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core) (0.7.0)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.7/dist-packages (from luma.core) (0.53.3)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.7/dist-packages (from luma.core) (0.4.1)
Requirement already satisfied: deprecated in /usr/local/lib/python3.7/dist-packages (from luma.core) (1.2.13)
Installing collected packages: luma.core, luma.led-matrix
Successfully installed luma.core-2.3.1 luma.led-matrix-1.5.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
pi@raspberrypi:~/luma.led_matrix $ sudo python3 examples/matrix_demo.py
Traceback (most recent call last):
  File "examples/matrix_demo.py", line 122, in <module>
    demo(args.cascaded, args.block_orientation, args.rotate, args.reverse_order)
  File "examples/matrix_demo.py", line 22, in demo
    rotate=rotate or 0, blocks_arranged_in_reverse_order=inreverse)
  File "/usr/local/lib/python3.7/dist-packages/luma/led_matrix/device.py", line 89, in __init__
    self.clear()
  File "/usr/local/lib/python3.7/dist-packages/luma/core/mixin.py", line 46, in clear
    self.display(Image.new(self.mode, self.size))
AttributeError: module 'PIL.Image' has no attribute 'new'
```

unitythemaker, 2021-11-06:
I think this problem is not luma specific. It's more like about Pillow. I'll try to research about it.

unitythemaker, 2021-11-06:
I ran the library on my RPi 3B without errors. The problem is related to Pi Zero. Can you make it run on Pi Zero? TY anyways.

thijstriemstra, 2021-11-06:
Hmm, I doubt that. What python version are you using? Same version on both RPi 3B and Pi Zero? What operating system (version)?

unitythemaker, 2021-11-06:
**Common**
```
Raspbian Lite (the versions are same because installed w/ RPi Imager and the image was cached and didn't redownload therefore the OS' are same.)

pi@raspberrypi:~ $ python3 --version
Python 3.7.3

Installed pip via sudo apt install python3-pip.
Ran apt upgrade before installing & trying these.
```

**RPi3B**
```
pi@raspberrypi:~ $ pip3 --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
# Notice the armv7l may it can help
```

**RPi Zero WH**
```
pi@raspberrypi:~ $ pip3 --version
pip 21.3.1 from /usr/local/lib/python3.7/dist-packages/pip (python 3.7)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux
# Notice the armv6l may it can help
```

thijstriemstra, 2021-11-06:
some ideas:

- try upgrading pip to 21.3.1 on rpi3b, see if it also breaks stuff there
- try installing a different version of Pillow (e.g. `pip install Pillow==8.2.0`)
- install `virtualenvwrapper` so you can stop using `sudo` and work in a virtual environment instead

unitythemaker, 2021-11-06:
* Upgrading pip and reinstalling packages didn't make any change on the both devices.
* Installing a different version made the Pi Zero run the code and there is flashes on the matrix. But it doesn't work or it works, I can't read what it writes.
* that was the first thing i tried but after not working i wanted to use ``sudo`` after not seeing any change

The matrix doesn't change it's initial shape and the code doesn't give any error also shows logs like
```
Created device
MAX7219 LED Matrix Demo
```
<details>
  <summary>Click to view image</summary>

[image]
</details>

However I can see the brightness of leds are changing slightly. I saw some issues about it like #102.
* checked the wiring and it's connected to the correct pins. (input ones including DIN)
* tried both 3.3V & 5V (no change)
* tried rebooting/re-plugging power adapter

Also I've newly noticed that the boards in the images are not same with mine. Mine have a MAX7219CNG integrated. I don't know if that will make any change about that problem but it worked with RPi3B first. Now it doesn't even if I re-install the OS and try the same things on RPi3B.

(I'm running the code [examples/matrix_demo.py](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py))

unitythemaker, 2021-11-06:
I've tried running this matrix with Arduino UNO and still no change. It seems like the problem is way more different. TYSM for helping me!

closed 2021-11-06 by unitythemaker

unitythemaker, 2021-11-07:
I found the problem. The integrated was connected incorrectly.

[image]
[image]
[image]

thijstriemstra, 2021-11-09:
> I found the problem. 

nice!

unitythemaker, 2021-11-09:
Oh, ok. As you can see in the images: The integrated is attached onto a black header. However in the PCB, it shows it's correct direction but it is soldered in the wrong direction. Therefore you attach the integrated by looking at the header's side that has a semicircle. Which is actually the wrong direction according to PCB.