rm-hull/luma.led_matrix #260: Installation on Raspberry Pi OS Lite (bullseye) aborts with an error

arevi9176, 2021-11-12:
First of all thanks for all your effort! Your work is greatly appreciated!

I've been using the luma.led_matrix package for quite a while without any issues on an RPI 3B with Raspberry Pi OS Lite (**buster**).
Now I tried the latest Raspberry Pi OS Lite (**bullseye**) release (same hardware,  setup from scratch).
Unfortunately the installation of  luma.led_matrix package aborts with an error. Seems  there is a problem with ws2812.

Any ideas?

#### Type of Raspberry Pi
Raspberry Pi 3 Model B V1.2

#### Linux Kernel version
Linux 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
Raspbian GNU/Linux 11 (bullseye)

Python 3.9.2

#### Expected behaviour
successful installation with

`sudo python3 -m pip install --upgrade luma.led_matrix`

#### Actual behaviour

`sudo usermod -a -G spi,gpio pi`
`sudo apt install build-essential python3-dev python3-pip libfreetype6-dev libjpeg-dev libopenjp2-7 libtiff5`
`sudo -H pip install --upgrade --ignore-installed pip setuptools`

runs without error, but ...

`sudo python3 -m pip install --upgrade luma.led_matrix`

gives 

_ERROR: Failed building wheel for ws2812_

see details here: https://pastebin.com/raw/TRquDVmd

thijstriemstra, 2021-11-12:
it's a hack but removing that library from setup.cfg (download tarball or do git checkout of source) and running pip install -e . should workaround the issue.

michaelWuensch, 2021-11-12:
I am having the exact same issue.
What does "hack" mean? Is something not going to work?

thijstriemstra, 2021-11-12:
apparently that ws2812 package is broken, if you don't use it, then use instructions i gave to skip installation of that 3rd party package.

michaelWuensch, 2021-11-12:
Ah ok,
I just use the MAX7219 so that should work.
I removed it, ran
sudo python3 setup.py install
and it installed without an error. Thanks!

arevi9176, 2021-11-13:
Thx for the quick response - works for me too!

thijstriemstra, 2021-11-13:
> due to some problems with setting up RPi.GPIO 

Still have those error messages?

thijstriemstra, 2021-11-14:
> due to some problems with setting up RPi.GPIO

I also encountered the issue with RPi.GPIO failing on Python 3.9 with bullseye. Use pre-release version of RPi.GPIO instead:

```
pip install RPi.GPIO==0.7.1a4
```

arevi9176, 2021-11-14:
confirmed - v0.7.1a4 runs without error ...

J0eK1ng, 2021-11-14:
> it's a hack but removing that library from setup.cfg (download tarball or do git checkout of source) and running pip install -e . should workaround the issue.

I'm also having the same issue with "bullseye". Can you please tell me what lines I should remove from the setup.cfg.
Thanks

thijstriemstra, 2021-11-14:
Get rid of the `ws2812; platform_machine=="armv7l" and platform_system=="Linux"` line in the `install_requires` section of [setup.cfg](https://github.com/rm-hull/luma.led_matrix/blob/master/setup.cfg)

Raspipython, 2021-11-20:
Hi.
Sorry, but i created an extra issue for this toppic because i havnt Seen this issue. 
I have the same error but i dont know how to remove the ws2812 library. Can anyone help me? 
Sorry for my bad englisch 😅

Raspipython, 2021-11-20:
Sorry, but i created an extra issue for this toppic, because i havnt Seen this issue. Im new at the raspberry pi and github, but i have the same error. I dont know how to remove the ws2812 library. @thijstriemstra described it, but i dont know,how to download the tarball. 
I hope, anyone can help me.
Sorry for my Bad english😅

thijstriemstra, 2021-11-20:
- Download source: https://github.com/rm-hull/luma.led_matrix/archive/refs/heads/master.zip
- Unzip it
- cd luma.led_matrix-master/
- Open setup.cfg and remove `ws2812; platform_machine=="armv7l" and platform_system=="Linux"` line in that file
- `pip install -e .` (or `sudo pip install -e .`)

Raspipython, 2021-11-20:
I did this , and now my raspberry pi says:
Usage:   
  pip install [options] <requirement specifier> [package-index-options] ...
  pip install [options] -r <requirements file> [package-index-options] ...
  pip install [options] [-e] <vcs project url> ...
  pip install [options] [-e] <local project path> ...
  pip install [options] <archive url/path> ...

-e option requires 1 argument
what to do next ?

thijstriemstra, 2021-11-20:
you forgot the `.` at the end. Command is:

```
pip install -e .
```

Raspipython, 2021-11-20:
now this appears:
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi
ERROR: file:///home/pi does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

thijstriemstra, 2021-11-20:
Try `sudo pip3 install -e .` or `pip3 install -e .`, Python 2 is not supported.

Raspipython, 2021-11-20:
Now the same error appears again:
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi
ERROR: file:///home/pi does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

thijstriemstra, 2021-11-20:
Please copy/paste all steps you performed and paste it on http://www.pastebin.com and link it here.

Raspipython, 2021-11-20:
here:
https://pastebin.com/Z9KBq8rS

thijstriemstra, 2021-11-20:
you wrote:

```
first i unpacked the zip at pi.
then i typed this:
```

but you forgot to:

```
cd luma.led_matrix-master/
```

Raspipython, 2021-11-20:
Sorry i forgot to write that i have deleted the code

Raspipython, 2021-11-21:
And what means the cd before luma.led_matrix-master?

Raspipython, 2021-11-21:
oh sorry i didnt noticed cd. now iti works! Thanks a lot!

thijstriemstra, 2021-11-27:
So the `ws2812` package's last release was on Oct 7, 2014: https://pypi.org/project/ws2812/

It's homepage is listed as (a redirect to) https://github.com/pimoroni/unicorn-hat

The documentation there calls it the `unicornhat` package:

```
sudo pip3 install unicornhat
```

but also mentions it requires the `rpi_ws281x` library:

> Note this library requires the rpi_ws281x Python library which you can find here: https://github.com/pimoroni/rpi_ws281x-python

luma.led_matrix currently requires the following libraries:

```
install_requires =
    luma.core>=2.2.0
    ws2812; platform_machine=="armv7l" and platform_system=="Linux"
    rpi_ws281x; platform_machine=="armv7l" and platform_system=="Linux"
```

I can only find an import refererencing the `rpi_ws281x` library in the [source code](https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py):

```python
    def __ws281x__(self):
        import _rpi_ws281x
        return _rpi_ws281x
```

I also did a search for `unicornhat` imports in luma.led_matrix but couldn't find any.

This all leads me to believe that the ws2812 library is not needed (fortunately). I will create a PR that removes the dependency.

linuxminty, 2021-12-02:
I removed `ws2812 `from the setup.cfg file in the luma.led_matrix directory on my Raspberry Pi and then ran `pip install -e .` This resulted in the following output:

```
pi@raspberrypi-4B:~/luma.led_matrix $ pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmp96j6zc13
       cwd: /home/pi/luma.led_matrix
  Complete output (51 lines):
  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/packaging/requirements.py", line 102, in __init__
      req = REQUIREMENT.parseString(requirement_string)
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/pyparsing.py", line 1654, in parseString
      raise exc
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/pyparsing.py", line 1644, in parseString
      loc, tokens = self._parse( instring, 0 )
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/pyparsing.py", line 1402, in _parseNoCache
      loc,tokens = self.parseImpl( instring, preloc, doActions )
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/pyparsing.py", line 3417, in parseImpl
      loc, exprtokens = e._parse( instring, loc, doActions )
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/pyparsing.py", line 1406, in _parseNoCache
      loc,tokens = self.parseImpl( instring, preloc, doActions )
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/pyparsing.py", line 3205, in parseImpl
      raise ParseException(instring, loc, self.errmsg, self)
  pkg_resources._vendor.pyparsing.ParseException: Expected stringEnd (at char 27), (line:1, col:28)
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
      return self._get_build_requires(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 143, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 7, in <module>
      setuptools.setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 152, in setup
      _install_setup_requires(attrs)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 145, in _install_setup_requires
      dist.parse_config_files(ignore_option_errors=True)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 801, in parse_config_files
      self._finalize_requires()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 528, in _finalize_requires
      self._move_install_requirements_markers()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 567, in _move_install_requirements_markers
      inst_reqs = list(pkg_resources.parse_requirements(spec_inst_reqs))
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/__init__.py", line 3098, in parse_requirements
      yield Requirement(line)
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/__init__.py", line 3108, in __init__
      super(Requirement, self).__init__(requirement_string)
    File "/usr/local/lib/python3.9/dist-packages/pkg_resources/_vendor/packaging/requirements.py", line 104, in __init__
      raise InvalidRequirement(
  pkg_resources.extern.packaging.requirements.InvalidRequirement: Parse error at "'and plat'": Expected stringEnd
  ----------------------------------------
WARNING: Discarding file:///home/pi/luma.led_matrix. Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmp96j6zc13 Check the logs for full command output.
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmp96j6zc13 Check the logs for full command output.
```
I don't know how to interpret the warning and error messages. How to proceed to get luma.led_matrix to work with my ZeroSeg 7-Segment display and the "Bullseye" install? Help would be much appreciated. Thank you.

linuxminty, 2021-12-02:
Thank you for your reply. Here's the requested output from setup.cfg:
```
pi@raspberrypi-4B:~/luma.led_matrix $ cat setup.cfg
[metadata]
name = luma.led_matrix
version = attr: luma.led_matrix.__version__
description = A library to drive a MAX7219 LED serializer (using SPI) and WS2812 NeoPixels (using DMA)
long_description = file: README.rst, CONTRIBUTING.rst, CHANGES.rst
long_description_content_type = text/x-rst
keywords = raspberry pi, rpi, led, max7219, matrix, seven segment, 7 segment, neopixel, neosegment, ws2812, ws281x, apa102, unicorn-phat, unicorn-hat, unicorn-hat-hd
author = Richard Hull
author_email = richard.hull@destructuring-bind.org
url = https://github.com/rm-hull/luma.led_matrix
license = MIT
classifiers =
    License :: OSI Approved :: MIT License
    Development Status :: 5 - Production/Stable
    Intended Audience :: Education
    Intended Audience :: Developers
    Topic :: Education
    Topic :: System :: Hardware
    Topic :: System :: Hardware :: Hardware Drivers
    Programming Language :: Python :: 3
    Programming Language :: Python :: 3.6
    Programming Language :: Python :: 3.7
    Programming Language :: Python :: 3.8
    Programming Language :: Python :: 3.9

[options]
zip_safe = False
packages = find:
python_requires = >=3.6, <4
namespace_packages = luma
install_requires =
    luma.core>=2.2.0
    platform_machine=="armv7l" and platform_system=="Linux"
    rpi_ws281x; platform_machine=="armv7l" and platform_system=="Linux"
tests_require =
    pytest
    pytest-cov
    pytest-timeout

[options.extras_require]
docs = sphinx>=1.5.1
qa = flake8; rstcheck
test = pytest; pytest-cov; pytest-timeout

[bdist_wheel]
universal = 1

[flake8]
ignore = E126, E127, E128, E241, E402, E501, E731, E741
exclude =
    .tox,
    # No need to traverse our git directory
    .git,
    .vscode,
    # There's no value in checking cache directories
    __pycache__,
    doc,
    build,
    dist

```

thijstriemstra, 2021-12-02:
replace

```
install_requires =
    luma.core>=2.2.0
    platform_machine=="armv7l" and platform_system=="Linux"
    rpi_ws281x; platform_machine=="armv7l" and platform_system=="Linux"
```

with

```
install_requires =
    luma.core>=2.2.0
    rpi_ws281x; platform_machine=="armv7l" and platform_system=="Linux"
```

linuxminty, 2021-12-03:
Hello Thijs--Success!!! Your guidance worked and the ZeroSeg is again displaying time and date on the Raspberry Pi with "Bullseye" installed. Thank you so much for taking the time to get this working for me!

One additional question, if I may: After making your suggested changes to setup.cnfg, I ran `pip install -e . `again and got the following:
```
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting rpi-ws281x
  Downloading https://www.piwheels.org/simple/rpi-ws281x/rpi_ws281x-4.3.1-cp39-cp39-linux_armv7l.whl (117 kB)
     |████████████████████████████████| 117 kB 333 kB/s            
Collecting luma.core>=2.2.0
  Downloading https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
     |████████████████████████████████| 71 kB 107 kB/s            
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.4.1)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.7.0)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.13)
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (5.4.2)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.53.3)
Requirement already satisfied: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (8.1.2)
Requirement already satisfied: wrapt<2,>=1.10 in /usr/lib/python3/dist-packages (from deprecated->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.12.1)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5b0)
Installing collected packages: rpi-ws281x, luma.core, luma.led-matrix
  Running setup.py develop for luma.led-matrix
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/luma.led_matrix/setup.py'"'"'; __file__='"'"'/home/pi/luma.led_matrix/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix=
         cwd: /home/pi/luma.led_matrix/
    Complete output (32 lines):
    running develop
    /usr/local/lib/python3.9/dist-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    WARNING: The user site-packages directory is disabled.
    /usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    error: can't create or remove files in install directory
    
    The following error occurred while trying to add or remove files in the
    installation directory:
    
        [Errno 13] Permission denied: '/usr/local/lib/python3.9/dist-packages/test-easy-install-4345.write-test'
    
    The installation directory you specified (via --install-dir, --prefix, or
    the distutils default setting) was:
    
        /usr/local/lib/python3.9/dist-packages/
    
    Perhaps your account does not have write access to this directory?  If the
    installation directory is a system-owned directory, you may need to sign in
    as the administrator or "root" account.  If you do not have administrative
    access to this machine, you may wish to choose a different installation
    directory, preferably one that is listed in your PYTHONPATH environment
    variable.
    
    For information on other options, you may wish to consult the
    documentation at:
    
      https://setuptools.pypa.io/en/latest/deprecated/easy_install.html
    
    Please make the appropriate changes for your system and try again.
    
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/luma.led_matrix/setup.py'"'"'; __file__='"'"'/home/pi/luma.led_matrix/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix= Check the logs for full command output.
```
[... cut ...] e to "permission denied," I then tried `sudo pip install -e .`
and it worked. My only concern is the concluding warning of that successful output:

`WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
`

Does this pose a problem for future use of the system package manager?

Again, thank you very much for your excellent support!

thijstriemstra, 2021-12-03:
> Does this pose a problem for future use of the system package manager?

No, don't worry about it.

radkechr, 2021-12-20:
> Remove the `ws2812` library from setup.cfg (download tarball or do git checkout of source) and running `pip install -e .` should workaround the issue.

After doing this, I encounter "ModuleNotFoundError: No module named 'luma.led_matrix'" error

joedellerfc, 2021-12-22:
I'm having the same trouble as radkechr,  I modified the setup.cfg, ran pip install -e . which ran OK, then tried to run an example and get the ModuleNotFoundError on the import luma.led_matrix

radkechr, 2021-12-22:
> I'm having the same trouble as radkechr, I modified the setup.cfg, ran pip install -e . which ran OK, then tried to run an example and get the ModuleNotFoundError on the import luma.led_matrix

Joe, to get around it I just used the previous version of raspi os (“buster”), and the package installed per the instructions, and you don’t need to do any of this.

nsauter, 2021-12-26:
I have the same issue as @joedellerfc & @radkechr, but installing buster is no real option since this is a 'old' os. Does anybody now how to fix that?

thijstriemstra, 2021-12-26:
> then tried to run an example and get the ModuleNotFoundError on the import luma.led_matrix

we need command(s) you ran and full traceback/error message in order to help. going back to buster is not an option imo.

nsauter, 2021-12-26:
@thijstriemstra Thats the error:

This comes from a fresh installed buster (i wanted to double check if its working with buster). Install went without errors but executing this example breaks
```
pi@raspberrypi:~/luma.led_matrix $ python examples/matrix_demo.py
Traceback (most recent call last):
  File "examples/matrix_demo.py", line 10, in <module>
    from luma.led_matrix.device import max7219
ImportError: No module named luma.led_matrix.device
```

joedellerfc, 2021-12-29:
> Joe, to get around it I just used the previous version of raspi os (“buster”), and the package installed per the instructions, and you don’t need to do any of this.

Thanks for the tip, will give that a go.

thijstriemstra, 2021-12-29:
@nsauter try `python --version` and make sure it's python 3. Make sure to install it using pip for python3, e.g. `pip3`.

nsauter, 2021-12-29:
There is only python3 installed in newer os Images since python2 is obsolete.

thijstriemstra, 2021-12-29:
I don't know how you installed luma.led_matrix, what the output was etc.

closed 2021-12-29 by thijstriemstra

suronw, 2022-01-03:
I'm having the same issue. Here are my steps and error message:

git clone https://github.com/rm-hull/luma.led_matrix.git
cd luma.led_matrix
sudo pip3 install -e .

error message:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/Downloads/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpmci60wz4
       cwd: /home/pi/Downloads/luma.led_matrix
  Complete output (48 lines):
  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 419, in _parse_attr
      return getattr(StaticModule(module_name), attr_name)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 27, in __init__
      with open(spec.origin) as strm:
  AttributeError: 'NoneType' object has no attribute 'origin'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
      return self._get_build_requires(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 143, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 7, in <module>
      setuptools.setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 155, in setup
      return distutils.core.setup(**attrs)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/core.py", line 122, in setup
      dist.parse_config_files()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 804, in parse_config_files
      parse_configuration(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 158, in parse_configuration
      meta.parse()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 498, in parse
      section_parser_method(section_options)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 469, in parse_section
      self[name] = value
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 222, in __setitem__
      value = parser(value)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 599, in _parse_version
      version = self._parse_attr(value, self.package_dir)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 422, in _parse_attr
      module = importlib.import_module(module_name)
    File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
      return _bootstrap._gcd_import(name[level:], package, level)
    File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
    File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
    File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
  ModuleNotFoundError: No module named 'luma.led_matrix'
  ----------------------------------------
WARNING: Discarding file:///home/pi/Downloads/luma.led_matrix. Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpmci60wz4 Check the logs for full command output.
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpmci60wz4 Check the logs for full command output.

thijstriemstra, 2022-01-03:
@suronw What version of Python? Did you compile it from source? What version of pip (`pip --version`)?

suronw, 2022-01-04:
pip 21.3.1 from /usr/local/lib/python3.9/dist-packages/pip (python 3.9)

tired running 
python3 -m pip install --upgrade setuptools
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in /usr/local/lib/python3.9/dist-packages (60.2.0)

same version

RPI-first, 2022-01-05:
Dear All,
anybody solved this installtion issue? I have the issue any more with RPI4.

Executed Commands:
sudo apt install perl libdevice-serialport-perl libio-socket-ssl-perl libwww-perl libxml-simple-perl
sudo apt-get install libdatetime-perl
sudo usermod -a -G spi,gpio pi
sudo apt install build-essential python3-dev python3-pip libfreetype6-dev libjpeg-dev libopenjp2-7 libtiff5
sudo -H pip install --upgrade --ignore-installed pip setuptools
git clone https://github.com/rm-hull/luma.led_matrix.git
cd luma.led_matrix
# Remove "ws2812" and "platform_machine=="armv7l" and platform_system=="Linux"  in der setup.cfg Datei
sudo pip install -e .

python matrix_demo.py 
          Traceback (most recent call last):
            File "/home/pi/luma.led_matrix/examples/matrix_demo.py", line 10, in <module>
              from luma.led_matrix.device import max7219
          ModuleNotFoundError: No module named 'luma.led_matrix'

pip3 list | grep luma
            luma.core         2.3.1

Could anyone help me?
(I wounder that pip3 list luma.led_matrix missing)

Thank you very much!

RPI-first, 2022-01-05:
Dear All,
anybody solved this installtion issue? I have the issue any more with RPI4.

Executed Commands:
sudo apt install perl libdevice-serialport-perl libio-socket-ssl-perl libwww-perl libxml-simple-perl
sudo apt-get install libdatetime-perl
sudo usermod -a -G spi,gpio pi
sudo apt install build-essential python3-dev python3-pip libfreetype6-dev libjpeg-dev libopenjp2-7 libtiff5
sudo -H pip install --upgrade --ignore-installed pip setuptools
git clone https://github.com/rm-hull/luma.led_matrix.git
cd luma.led_matrix

Remove "ws2812" and "platform_machine=="armv7l" and platform_system=="Linux" in der setup.cfg Datei
sudo pip install -e .

python matrix_demo.py
Traceback (most recent call last):
File "/home/pi/luma.led_matrix/examples/matrix_demo.py", line 10, in
from luma.led_matrix.device import max7219
ModuleNotFoundError: No module named 'luma.led_matrix'

pip3 list | grep luma
luma.core 2.3.1

Could anyone help me?
(I wounder that pip3 list luma.led_matrix missing)

Thank you very much!

jarleven, 2022-01-09:
Same here on Bullseye

pi@raspberrypi:~ $ cd luma.led_matrix/
pi@raspberrypi:~/luma.led_matrix $ vi setup.cfg
pi@raspberrypi:~/luma.led_matrix $ pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpsy_4173u
       cwd: /home/pi/luma.led_matrix
  Complete output (48 lines):
  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 419, in _parse_attr
      return getattr(StaticModule(module_name), attr_name)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 27, in __init__
      with open(spec.origin) as strm:
  AttributeError: 'NoneType' object has no attribute 'origin'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
      return self._get_build_requires(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 143, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 7, in <module>
      setuptools.setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 155, in setup
      return distutils.core.setup(**attrs)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/core.py", line 122, in setup
      dist.parse_config_files()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 804, in parse_config_files
      parse_configuration(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 158, in parse_configuration
      meta.parse()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 498, in parse
      section_parser_method(section_options)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 469, in parse_section
      self[name] = value
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 222, in __setitem__
      value = parser(value)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 599, in _parse_version
      version = self._parse_attr(value, self.package_dir)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 422, in _parse_attr
      module = importlib.import_module(module_name)
    File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
      return _bootstrap._gcd_import(name[level:], package, level)
    File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
    File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
    File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
  ModuleNotFoundError: No module named 'luma.led_matrix'
  ----------------------------------------
WARNING: Discarding file:///home/pi/luma.led_matrix. Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_proce                                                             ss/_in_process.py get_requires_for_build_wheel /tmp/tmpsy_4173u Check the logs for full command output.
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel                                                              /tmp/tmpsy_4173u Check the logs for full command output.
pi@raspberrypi:~/luma.led_matrix $

jarleven, 2022-01-09:
sudo and verbose build

```bash
pi@raspberrypi:~/luma.led_matrix $ sudo pip install -vv -e .
Using pip 21.3.1 from /usr/local/lib/python3.9/dist-packages/pip (python 3.9)
Non-user install because site-packages writeable
Created temporary directory: /tmp/pip-ephem-wheel-cache-d19vu8vi
Created temporary directory: /tmp/pip-req-tracker-71qpdrqe
Initialized build tracking at /tmp/pip-req-tracker-71qpdrqe
Created build tracker: /tmp/pip-req-tracker-71qpdrqe
Entered build tracker: /tmp/pip-req-tracker-71qpdrqe
Created temporary directory: /tmp/pip-install-fxkr0127
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Added file:///home/pi/luma.led_matrix to build tracker '/tmp/pip-req-tracker-71qpdrqe'
  Created temporary directory: /tmp/pip-build-env-hx1q4235
  Created temporary directory: /tmp/pip-standalone-pip-wnt25btp
  Running command /usr/bin/python3 /tmp/pip-standalone-pip-wnt25btp/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-hx1q4235/overlay --no-warn-script-location -v --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools >= 40.6.0' wheel
  Using pip 21.3.1 from /tmp/pip-standalone-pip-wnt25btp/__env_pip__.zip/pip (python 3.9)
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Using cached https://www.piwheels.org/simple/setuptools/setuptools-60.5.0-py3-none-any.whl (958 kB)
  Collecting wheel
    Using cached https://www.piwheels.org/simple/wheel/wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Installing collected packages: wheel, setuptools
    Creating /tmp/pip-build-env-hx1q4235/overlay/bin
    changing mode of /tmp/pip-build-env-hx1q4235/overlay/bin/wheel to 755
  Successfully installed setuptools-60.5.0 wheel-0.37.1
  WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
  Installing build dependencies ... done
  Running command /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py _supported_features /tmp/tmp1s39fhgc
  Checking if build backend supports build_editable ... done
  Running command /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpu1kbo0xs
  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 419, in _parse_attr
      return getattr(StaticModule(module_name), attr_name)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 27, in __init__
      with open(spec.origin) as strm:
  AttributeError: 'NoneType' object has no attribute 'origin'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
      return self._get_build_requires(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 143, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 7, in <module>
      setuptools.setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 155, in setup
      return distutils.core.setup(**attrs)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/core.py", line 122, in setup
      dist.parse_config_files()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 804, in parse_config_files
      parse_configuration(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 158, in parse_configuration
      meta.parse()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 498, in parse
      section_parser_method(section_options)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 469, in parse_section
      self[name] = value
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 222, in __setitem__
[... 36 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/resolution/resolvelib/candidates.py", line 316, in _prepare_distribution
    return self._factory.preparer.prepare_editable_requirement(self._ireq)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/operations/prepare.py", line 599, in prepare_editable_requirement
    dist = _get_prepared_distribution(
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/operations/prepare.py", line 58, in _get_prepared_distribution
[... cut ...]
    raise InstallationSubprocessError(proc.returncode, command_desc)
pip._internal.exceptions.InstallationSubprocessError: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpu1kbo0xs Check the logs for full command output.
Removed file:///home/pi/luma.led_matrix from build tracker '/tmp/pip-req-tracker-71qpdrqe'
Removed build tracker: '/tmp/pip-req-tracker-71qpdrqe'
pi@raspberrypi:~/luma.led_matrix $
```

jarleven, 2022-01-09:
It works when using virtual environment.

```bash
pi@raspberrypi:~ $ sudo apt-get install python3-venv
pi@raspberrypi:~ $ python3 -m venv matrix-env

pi@raspberrypi:~ $ source matrix-env/bin/activate

(matrix-env) pi@raspberrypi:~ $ git clone https://github.com/rm-hull/luma.led_matrix.git

(matrix-env) pi@raspberrypi:~ $ cd luma.led_matrix/

(matrix-env) pi@raspberrypi:~/luma.led_matrix $ vi setup.cfg

(matrix-env) pi@raspberrypi:~/luma.led_matrix $ pip install -vvv -e .

(matrix-env) pi@raspberrypi:~/luma.led_matrix $ python3 examples/matrix_demo.py
Created device
MAX7219 LED Matrix Demo
```

jarleven, 2022-01-09:
```bash
(matrix-env) pi@raspberrypi:~/luma.led_matrix $ git diff
diff --git a/setup.cfg b/setup.cfg
index 3aeee44..23fa5e2 100644
--- a/setup.cfg
+++ b/setup.cfg
@@ -31,7 +31,6 @@ python_requires = >=3.6, <4
 namespace_packages = luma
 install_requires =
     luma.core>=2.2.0
-    rpi_ws281x; platform_machine=="armv7l" and platform_system=="Linux"
 tests_require =
     pytest
     pytest-cov
(matrix-env) pi@raspberrypi:~/luma.led_matrix $
```

jarleven, 2022-01-09:
```bash
(matrix-env) pi@raspberrypi:~/luma.led_matrix $ python3 -m pip list
Package         Version Location
--------------- ------- ------------------------
cbor2           5.4.2
Deprecated      1.2.13
luma.core       2.3.1
luma.led-matrix 1.6.0   /home/pi/luma.led_matrix
Pillow          9.0.0
pip             20.3.4
pkg-resources   0.0.0
pyftdi          0.53.3
pyserial        3.5
pyusb           1.2.1
RPi.GPIO        0.7.0
setuptools      44.1.1
smbus2          0.4.1
spidev          3.5
wrapt           1.13.3
```

pi@raspberrypi:~ $ python3 -m pip list
Package       Version
------------- ---------
cbor2         5.4.2
certifi       2020.6.20
chardet       4.0.0
colorzero     1.1
Deprecated    1.2.13
distro        1.5.0
gpiozero      1.6.2
idna          2.10
luma.core     2.3.1
Pillow        9.0.0
pip           21.3.1
pyftdi        0.53.3
pyserial      3.5
python-apt    2.2.1
pyusb         1.2.1
requests      2.25.1
RPi.GPIO      0.7.0
setuptools    60.5.0
six           1.16.0
smbus2        0.4.1
spidev        3.5
SQLAlchemy    1.4.29
ssh-import-id 5.10
urllib3       1.26.5
wheel         0.34.2
wrapt         1.13.3
```bash
pi@raspberrypi:~ $ python3 -m pip list
Package       Version
------------- ---------
cbor2         5.4.2
certifi       2020.6.20
chardet       4.0.0
colorzero     1.1
Deprecated    1.2.13
distro        1.5.0
gpiozero      1.6.2
idna          2.10
luma.core     2.3.1
Pillow        9.0.0
pip           21.3.1
pyftdi        0.53.3
pyserial      3.5
python-apt    2.2.1
pyusb         1.2.1
requests      2.25.1
RPi.GPIO      0.7.0
setuptools    60.5.0
six           1.16.0
smbus2        0.4.1
spidev        3.5
SQLAlchemy    1.4.29
ssh-import-id 5.10
urllib3       1.26.5
wheel         0.34.2
wrapt         1.13.3
```

```bash
pi@raspberrypi:~ $ sudo python3 -m pip list
Package       Version
------------- ---------
cbor2         5.4.2
certifi       2020.6.20
chardet       4.0.0
colorzero     1.1
Deprecated    1.2.13
distro        1.5.0
gpiozero      1.6.2
idna          2.10
luma.core     2.3.1
Pillow        9.0.0
pip           21.3.1
pyftdi        0.53.3
pyserial      3.5
python-apt    2.2.1
pyusb         1.2.1
requests      2.25.1
RPi.GPIO      0.7.0
setuptools    60.5.0
six           1.16.0
smbus2        0.4.1
spidev        3.5
SQLAlchemy    1.4.29
ssh-import-id 5.10
urllib3       1.26.5
wheel         0.34.2
wrapt         1.13.3
```

jarleven, 2022-01-09:
This is working for me on Bullseye. The last thing I had to add to my script was "export CFLAGS=-fcommon" as  specified by arevi9176

```bash
wget https://raw.githubusercontent.com/jarleven/flaskapp/a53e8440dbf7e674141aca3e6061c21b14f708a6/luma/setup_matrix.sh && chmod +x setup_matrix.sh && ./setup_matrix.sh
```

thijstriemstra, 2022-01-09:
> The last thing I had to add to my script was "export CFLAGS=-fcommon" as specified by arevi9176

This can also be fixed by using pre-release version of RPi.GPIO instead: `pip install RPi.GPIO==0.7.1a4`

PR #264 (2021-12-29): remove redundant ws2812 package
See https://github.com/rm-hull/luma.led_matrix/issues/260#issuecomment-980800315

fixes #260
commit 59cdec76 (2021-12-29): remove redundant ws2812 package (#264)

fixes #260