rm-hull/luma.led_matrix #263: problem installing luma.led

rcaspergit, 2021-11-27:
#### Type of Raspberry Pi

RPi3 model B version 1.2

#### Linux Kernel version
uname -a
Linux pi3 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

#### Expected behaviour

attempting to install  luma.led_matrix following https://luma-led-matrix.readthedocs.io/en/latest/install.html to experiment with an 8x8 led matrix tied to a MAX7219

#### Actual behaviour

Following the above instructions the install errors out

pip -V
pip 21.3.1 from /usr/local/lib/python3.9/dist-packages/pip (python 3.9)
pi@pi3:~/github $  sudo python3 -m pip install --upgrade luma.led_matrix
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: luma.led_matrix in /usr/local/lib/python3.9/dist-packages/luma.led_matrix-1.5.0-py3.9.egg (1.5.0)
Collecting luma.core>=2.2.0
  Using cached https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
Collecting ws2812
  Using cached ws2812-1.0.0.tar.gz (58 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: rpi_ws281x in /usr/local/lib/python3.9/dist-packages/rpi_ws281x-4.3.1-py3.9-linux-armv7l.egg (from luma.led_matrix) (4.3.1)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (3.5)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (0.53.3)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (0.4.1)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (0.7.0)
Requirement already satisfied: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (8.1.2)
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (5.4.2)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led_matrix) (1.2.13)
Collecting wrapt<2,>=1.10
  Downloading https://www.piwheels.org/simple/wrapt/wrapt-1.13.3-cp39-cp39-linux_armv7l.whl (71 kB)
     |████████████████████████████████| 71 kB 332 kB/s
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led_matrix) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led_matrix) (3.5b0)
Building wheels for collected packages: ws2812
  Building wheel for ws2812 (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4uaw5hzf/ws2812_101d675cb472415288c91b02340f5330/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4uaw5hzf/ws2812_101d675cb472415288c91b02340f5330/setup.py'"

many more errors and the the install script exits. I preformed all the requested updates prior to attempting the install,

thijstriemstra, 2021-11-27:
Duplicate of #260

closed 2021-11-27 by thijstriemstra

rcaspergit, 2021-11-27:
> Duplicate of #260

Thank you, that worked.