rm-hull/luma.led_matrix #266: Threading or Multiprocessing

J0eK1ng, 2022-01-02:
Hey guys,

I have a python script in the works and I'm able to send text to the LED matrix without issues, it's working great.

My question is. I'm sending so much text to be displayed that my Pi3 and python script must wait for all the text to be displayed before it can continue to process the rest of my script. Would it be possible to either run the LED matrix portion of the script in another Thread or Multiprocess?

Has this been done before or can it be done?

Thanks

Denis
(Sorry for the bad english it's my second language)

rm-hull, 2022-01-09:
without an example of what your script does i cannot say. you can run the display device in a separate thread, but would have to manage locking acccess yourself.

closed 2022-01-09 by rm-hull

J0eK1ng, 2022-01-09:
Ok Thanks for the reply. I wanted to know if it could be done. I will research and investigate how to do it. If there is any interest I can post back my findings.