rm-hull/luma.led_matrix #268: "Illegal instruction" error when importing max7219

iglennon-qualys, 2022-01-03:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

PiZero W

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

I would expect python3 to not terminate when I run the command

from luma.led_matrix.device import max7219

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

This is a direct copy/paste of the behaviour
```
pi@raspberrypi:~/luma.led_matrix/examples $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from luma.led_matrix.device import max7219
Illegal instruction
pi@raspberrypi:~/luma.led_matrix/examples $ echo $?
132
pi@raspberrypi:~/luma.led_matrix/examples $
```

thijstriemstra, 2022-01-03:
what version of luma.led_matrix is this? and did you compile python from source? can you paste output of `pip3 list`?

iglennon-qualys, 2022-01-03:
I'm going to close this, I opened it under the wrong Github account.  I will open another issue and paste the output requested

closed 2022-01-03 by iglennon-qualys

thijstriemstra, 2022-01-03:
Duplicate of #269