rm-hull/luma.led_matrix #269: "Illegal Instruction" error

ianglennon, 2022-01-03:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

PiZero W

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

I would expect python3 to not terminate when I run the command

from luma.led_matrix.device import max7219

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

```
pi@raspberrypi:~/luma.led_matrix/examples $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from luma.led_matrix.device import max7219
Illegal instruction
pi@raspberrypi:~/luma.led_matrix/examples $ echo $?
132
pi@raspberrypi:~/luma.led_matrix/examples $
```

Output of pip3 list

```
Package         Version
--------------- ---------
cbor2           5.4.2
certifi         2020.6.20
chardet         4.0.0
colorzero       1.1
Deprecated      1.2.13
distro          1.5.0
gpiozero        1.6.2
idna            2.10
luma.core       2.3.1
luma.led-matrix 1.5.0
Pillow          9.0.0
pip             20.3.4
pyftdi          0.53.3
pyserial        3.5
python-apt      2.2.1
pyusb           1.2.1
requests        2.25.1
RPi.GPIO        0.7.0
setuptools      52.0.0
six             1.16.0
smbus2          0.4.1
spidev          3.5
ssh-import-id   5.10
urllib3         1.26.5
wheel           0.34.2
wrapt           1.13.3
```

When running ```from PIL import Image``` I get the same error

thijstriemstra, 2022-01-03:
> When running from PIL import Image I get the same error

Can you try installing an older version of Pillow, e.g. `pip install "Pillow==8.4.0"`?

ianglennon, 2022-01-03:
I'm going to hold this until #271 gets resolved.  If I can't install and run it on a 3B+ then I pretty much have no hope of getting it going on the ZeroW

thijstriemstra, 2022-01-03:
Let me ask again; Can you try installing an older version of Pillow, e.g. pip install "Pillow==8.4.0"?

ianglennon, 2022-01-03:
THIS IS NOT THE SAME ERROR!  IT IS NOT EVEN THE SAME RPI!

I'm done.

thijstriemstra, 2022-01-03:
hey, im trying to help you.

ianglennon, 2022-01-03:
Yeah, I know, sorry for the unnecessary all-caps response.  I have quite literally been trying to get this going all day and have had zero luck.  I have tried everything that has been suggested on the other tickets, all resulting in some kind of failure.  I have simply lost patience with it.  This was supposed to be the final and easiest step in designing and building my own device, starting off with a simple 7-digit display with a MAX7219 driver.  I had my own PCB fabricated, I spent the morning soldering components to it, and I even got to see it light up.  After that, nothing else has gone right.

I'm not bothered about the Pi Zero not working, I only used it because my 3B+ was busy doing something else.  Now that it's available I want to focus on that instead.  That's the other ticket, btw... #270 

I'll re-open that one (I honestly didn't realise I could re-open a ticket) and we'll try again.

ianglennon, 2022-01-03:
I foolishly re-opened this one instead of #270.  I'm an idiot 🤣

closed 2022-01-03 by ianglennon

StefanSchroeder, 2022-01-17:
I observed the same error that @ianglennon described so well. I ran **pip install "Pillow==8.4.0"** as suggested. This fixes the problem. 👍