rm-hull/luma.led_matrix #270: Package fails to install via pip

ianglennon, 2022-01-03:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

Pi 3B+

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

Just trying to install the package

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

Basically, ws2812 is failing to install with a whole bunch of warnings and errors

Brace yourself, this is a looong output

```
pi@raspberrypi:~ $ sudo python3 -m pip install --upgrade luma.led_matrix
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting luma.led_matrix
  Downloading https://www.piwheels.org/simple/luma-led-matrix/luma.led_matrix-1.5.0-py2.py3-none-any.whl (19 kB)
Collecting luma.core>=1.14.0
  Downloading https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
     |████████████████████████████████| 71 kB 936 kB/s
Collecting ws2812
  Downloading ws2812-1.0.0.tar.gz (58 kB)
     |████████████████████████████████| 58 kB 913 kB/s
  Preparing metadata (setup.py) ... done
Collecting rpi-ws281x
  Downloading https://www.piwheels.org/simple/rpi-ws281x/rpi_ws281x-4.3.1-cp39-cp39-linux_armv7l.whl (117 kB)
     |████████████████████████████████| 117 kB 1.3 MB/s
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (0.4.1)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (0.7.0)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (1.2.13)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (3.5)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (0.53.3)
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (5.4.2)
Requirement already satisfied: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=1.14.0->luma.led_matrix) (8.1.2)
Requirement already satisfied: wrapt<2,>=1.10 in /usr/lib/python3/dist-packages (from deprecated->luma.core>=1.14.0->luma.led_matrix) (1.12.1)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi->luma.core>=1.14.0->luma.led_matrix) (3.5b0)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=1.14.0->luma.led_matrix) (1.2.1)
Building wheels for collected packages: ws2812
  Building wheel for ws2812 (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/setup.py'"'"'; __file__='"'"'/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-ok60wk8h
       cwd: /tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/
  Complete output (151 lines):
  running bdist_wheel
  running build
  running build_py
[... 289 lines cut ...]
      341 | static volatile unsigned int *gpio_reg;  // GPIO pin controller register set
          |                               ^~~~~~~~
    lib/ws2812-RPi.h:340:31: warning: ‘dma_reg’ defined but not used [-Wunused-variable]
      340 | static volatile unsigned int *dma_reg;  // DMA controller register set
          |                               ^~~~~~~
    lib/ws2812-RPi.h:339:31: warning: ‘clk_reg’ defined but not used [-Wunused-variable]
      339 | static volatile unsigned int *clk_reg;  // PWM clock manager register set
          |                               ^~~~~~~
    lib/ws2812-RPi.h:338:31: warning: ‘pwm_reg’ defined but not used [-Wunused-variable]
      338 | static volatile unsigned int *pwm_reg;  // PWM controller register set
          |                               ^~~~~~~
    lib/ws2812-RPi.h:336:17: warning: ‘virtbase’ defined but not used [-Wunused-variable]
      336 | static uint8_t *virtbase;     // Pointer to some virtual memory that will be allocated
          |                 ^~~~~~~~
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-armv7l-3.9/lib/ws2812-RPi.o build/temp.linux-armv7l-3.9/ws2812-RPi_wrap.o -Llib/ -L/usr/lib -o build/lib.linux-armv7l-3.9/_ws2812.cpython-39-arm-linux-gnueabihf.so
    /usr/bin/ld: build/temp.linux-armv7l-3.9/ws2812-RPi_wrap.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:415: multiple definition of `PWMWaveform'; build/temp.linux-armv7l-3.9/lib/ws2812-RPi.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:415: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/ws2812-RPi_wrap.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:410: multiple definition of `LEDBuffer'; build/temp.linux-armv7l-3.9/lib/ws2812-RPi.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:410: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/ws2812-RPi_wrap.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:407: multiple definition of `numLEDs'; build/temp.linux-armv7l-3.9/lib/ws2812-RPi.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:407: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/ws2812-RPi_wrap.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:398: multiple definition of `brightness'; build/temp.linux-armv7l-3.9/lib/ws2812-RPi.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:398: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/ws2812-RPi_wrap.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:335: multiple definition of `page_map'; build/temp.linux-armv7l-3.9/lib/ws2812-RPi.o:/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/lib/ws2812-RPi.h:335: first defined here
    collect2: error: ld returned 1 exit status
    error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/setup.py'"'"'; __file__='"'"'/tmp/pip-install-r3ys8oa0/ws2812_25925b960daf458295abdce83ced09f9/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-h416vp5n/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/ws2812 Check the logs for full command output.
```

thijstriemstra, 2022-01-03:
duplicate of #260

ianglennon, 2022-01-03:
So, as I was saying in #271 before I got all rude and unnecessarily angry...

I have followed the instructions in #260 and they did not work either.  Here's the output from the last thing I tried

```
pi@raspberrypi:~ $ git clone https://github.com/rm-hull/luma.led_matrix.git
Cloning into 'luma.led_matrix'...
remote: Enumerating objects: 2202, done.
remote: Counting objects: 100% (37/37), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 2202 (delta 10), reused 4 (delta 1), pack-reused 2165
Receiving objects: 100% (2202/2202), 11.34 MiB | 2.90 MiB/s, done.
Resolving deltas: 100% (1286/1286), done.
pi@raspberrypi:~ $ cd luma.led_matrix/
pi@raspberrypi:~/luma.led_matrix $ ls
CHANGES.rst  CONTRIBUTING.rst  doc  examples  LICENSE.rst  luma  MANIFEST.in  pyproject.toml  pytest.ini  README.rst  setup.cfg  setup.py  tests  tox.ini
pi@raspberrypi:~/luma.led_matrix $ vi setup.cfg
pi@raspberrypi:~/luma.led_matrix $ sudo pip3 install -e .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting luma.core>=2.2.0
  Downloading https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
     |████████████████████████████████| 71 kB 755 kB/s
Collecting rpi-ws281x
  Downloading https://www.piwheels.org/simple/rpi-ws281x/rpi_ws281x-4.3.1-cp39-cp39-linux_armv7l.whl (117 kB)
     |████████████████████████████████| 117 kB 1.9 MB/s
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (5.4.2)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.4.1)
Requirement already satisfied: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (8.1.2)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.7.0)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.53.3)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.13)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5)
Requirement already satisfied: wrapt<2,>=1.10 in /usr/lib/python3/dist-packages (from deprecated->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.12.1)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5b0)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.1)
Installing collected packages: rpi-ws281x, luma.core, luma.led-matrix
  Running setup.py develop for luma.led-matrix
Successfully installed luma.core-2.3.1 luma.led-matrix rpi-ws281x-4.3.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
pi@raspberrypi:~/luma.led_matrix $ cd examples/
pi@raspberrypi:~/luma.led_matrix/examples $ ./sevensegment_demo.py
Traceback (most recent call last):
  File "./sevensegment_demo.py", line 13, in <module>
    from luma.led_matrix.device import max7219
ImportError: No module named luma.led_matrix.device
```

ianglennon, 2022-01-03:
Oh, I see I can't re-open this ticket.  I'm guessing I could only re-open #269 because I closed it.  Can you re-open this one please?

thijstriemstra, 2022-01-03:
> Can you re-open this one please?

I will, once we resolve #269. I keep asking you to install older version of Pillow, but you ignore it. See #269 please.

ianglennon, 2022-01-03:
That was the Pi Zero, I'm not interested in pursuing that at all.  This is a different device, a Pi 3 B+, with a different problem.  The Pi Zero is now redployed doing something else so I won't be reattaching my device to it.

So, please let's just focus on *this* ticket.

thijstriemstra, 2022-01-03:
> pi@raspberrypi:~/luma.led_matrix/examples $ ./sevensegment_demo.py

Please run it with `python3 sevensegment_demo.py` instead and make sure that this `python3` your using has luma.led_matrix installed by checking it's listed in `python3 -m pip list`.

ianglennon, 2022-01-03:
I did try the example first, it gave the same error.  ```python3 -m pip list | grep luma``` only shows ```luma.core``` installed

```
pi@raspberrypi:~/luma.led_matrix $ python3 -m pip list | grep luma
luma.core         2.3.1
```

This is despite the message earier 

> Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.1)
> Installing collected packages: [...]

thijstriemstra, 2022-01-03:
> Please run it with python3 sevensegment_demo.py instead an 

And it's actually even in the instructions: https://github.com/rm-hull/luma.examples#running-the-examples

> This is despite the message earier

Ok, re-install it using `python3 -m pip install -e .`? It's probably the whole `sudo` thing that the docs instruct you to use..

ianglennon, 2022-01-03:
Thanks for re-opening this, by the way.  I'm sat here with the 2 x 7 segment displays attached to the Pi and glowing, hoping that I can get this working so I can get some actual control over it

thijstriemstra, 2022-01-03:
Python is notorious for these kind of installation issues. I would suggest trying to setup `virtualenvwrapper` so you can have much more control of your local python installation(s) and not touch the system.

ianglennon, 2022-01-03:
Oh.  Well that's disappointing, I was really hoping you would be able to help get this going without virtualenvwrapper.  I was attracted to this because I'm not confident enough to get that going and it wasn't referenced in the documentation so I thought it wasn't required.

thijstriemstra, 2022-01-03:
It's not needed, sounds like you're almost there without using virtualenvwrapper.

like I said:

> Ok, re-install it using python3 -m pip install -e .? It's probably the whole sudo thing that the docs instruct you to use..

ianglennon, 2022-01-03:
Oh, sorry, I missed that message.  I did try that earlier but I've tried it again, below is the error

```
pi@raspberrypi:~/luma.led_matrix $ python3 -m pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpieeg1wvo
       cwd: /home/pi/luma.led_matrix
  Complete output (48 lines):
  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 419, in _parse_attr
      return getattr(StaticModule(module_name), attr_name)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 27, in __init__
      with open(spec.origin) as strm:
  AttributeError: 'NoneType' object has no attribute 'origin'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
      return self._get_build_requires(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 143, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 7, in <module>
      setuptools.setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 155, in setup
      return distutils.core.setup(**attrs)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/core.py", line 122, in setup
      dist.parse_config_files()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/dist.py", line 804, in parse_config_files
      parse_configuration(
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 158, in parse_configuration
      meta.parse()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 498, in parse
      section_parser_method(section_options)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 469, in parse_section
      self[name] = value
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 222, in __setitem__
      value = parser(value)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 599, in _parse_version
      version = self._parse_attr(value, self.package_dir)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 422, in _parse_attr
      module = importlib.import_module(module_name)
    File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
      return _bootstrap._gcd_import(name[level:], package, level)
    File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
    File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
    File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
  ModuleNotFoundError: No module named 'luma.led_matrix'
  ----------------------------------------
WARNING: Discarding file:///home/pi/luma.led_matrix. Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpieeg1wvo Check the logs for full command output.
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.9/dist-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpieeg1wvo Check the logs for full command output.
```

thijstriemstra, 2022-01-03:
```
 Traceback (most recent call last):
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 419, in _parse_attr
      return getattr(StaticModule(module_name), attr_name)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/config.py", line 27, in __init__
      with open(spec.origin) as strm:
  AttributeError: 'NoneType' object has no attribute 'origin'
```

try updating setuptools? v60 something is the latest.

ianglennon, 2022-01-03:
setuptools is already at 60.2.0

```
pi@raspberrypi:~/luma.led_matrix $ python3 -m pip install --upgrade setuptools
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in /usr/local/lib/python3.9/dist-packages (60.2.0)
pi@raspberrypi:~/luma.led_matrix $ python3 -m pip list | grep setuptools
setuptools        60.2.0
```

jarleven, 2022-01-08:
I stumbeled into this thread googling for a solution.
I'm having issues with the ws2812 when running 

sudo python3 -m pip install --upgrade luma.led_matrix 

Pi3 and Pi4 have same issue when using Bullseye. On Buster everything works as expeced.

  ERROR: Failed building wheel for ws2812
  Running setup.py clean for ws2812
Failed to build ws2812
Installing collected packages: ws2812, luma.led-matrix
    Running setup.py install for ws2812 ... error
    ERROR: Command errored out with exit status 1:

Fishhooks1945, 2022-01-09:
Hey, I'm 12 this is my first time on GitHub. I am having these same errors.

- I edited setup.cfg and removed the ws2812 line per issue #260 
- Updated pip and setuptools
`rgbpi@rgbpi:~/luma.led_matrix-master $ python3 -m pip list
Package       Version
------------- ---------
cbor2         5.4.2
certifi       2020.6.20
chardet       4.0.0
colorzero     1.1
Deprecated    1.2.13
distro        1.5.0
gpiozero      1.6.2
gyp           0.1
idna          2.10
luma.core     2.3.1
Pillow        8.4.0
pip           21.3.1
pyftdi        0.53.3
pyserial      3.5
python-apt    2.2.1
pyusb         1.2.1
requests      2.25.1
RPi.GPIO      0.7.0
rpi-ws281x    4.3.1
setuptools    60.5.0
six           1.16.0
smbus2        0.4.1
spidev        3.5
ssh-import-id 5.10
ufw           0.36
urllib3       1.26.5
wheel         0.34.2
wrapt         1.13.3
`

Please help :) I promise to be very nice and grateful to the devs. ;)

ianglennon, 2022-01-09:
You both  need to raise your own issues, I'm about to close this as I switched to Arduino

closed 2022-01-09 by ianglennon

jarleven, 2022-01-09:
ianglennon I got it working as descripbed in :
https://github.com/rm-hull/luma.led_matrix/issues/260#issuecomment-1008312187

I also used the MD_MAX72xx libarary on Arduino, but I have changed to Luma as it is simple to do complex stuff on the Raspberry Pi.

rm-hull, 2022-01-09:
v1.6.1 has been released which no longer relies on ws2812

Fishhooks1945, 2022-01-09:
> You both need to raise your own issues, I'm about to close this as I switched to Arduino

my issue is the same one you are having.. i know i'm pretty new to github, but i'm pretty sure duplicating issues is frowned upon, just like your abrasive attitude. Jeex I hope everyone in the FOSS community isn't like you!

@rm-hull thank you for your work!!! So, should I just clone the repo and try installing again with `sudo pip3 install -e .` ?
THANK YOU for taking your time with me, i am VERY new at this

thijstriemstra, 2022-01-09:
> my issue is the same one you are having.. 

Agreed. No worries @Fishhooks1945

thijstriemstra, 2022-01-09:
>  So, should I just clone the repo and try installing again with sudo pip3 install -e . ?

You should be able to just use `sudo pip3 install luma.led_matrix` since we released a new version today.

Fishhooks1945, 2022-01-09:
> You should be able to just use `sudo pip3 install luma.led_matrix` since we released a new version today.

Wow that worked! My LED array isn't turning on but that is probably my fault somewhere... THANK YOU @thijstriemstra , @rm-hull and the rest of the devs!

jarleven, 2022-01-09:
> > You should be able to just use `sudo pip3 install luma.led_matrix` since we released a new version today.
> 
> Wow that worked! My LED array isn't turning on but that is probably my fault somewhere... THANK YOU @thijstriemstra , @rm-hull and the rest of the devs!

Did you enable the SPI interface and allow the pi user to use it ?

```
sudo raspi-config nonint do_spi 0
sudo usermod -a -G spi,gpio pi
```

Fishhooks1945, 2022-01-09:
yeah i did all that, but I first tried matrix_demo.py but that isn't for my ws2812 "unicorn" 8x8 array. When I ran neopixel_demo.py it worked great, because i already had GPIO18 hooked up to my matrix :)
[image]

I've been having fun on this all day, I made a smiley face with draw.point function.