rm-hull/luma.led_matrix #271: Issue #270 wasn't resolved by instructions in #260

ianglennon, 2022-01-03:
See #270 for the history (yes, I know it's a duplicate, but that didn't help)

Here's what I'm faced with now

Following the instructions in #260 didn't help at all, despite promising signs that it installed successfully.

```
pi@raspberrypi:~/luma.led_matrix $ sudo -H pip3 install -e .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting luma.core>=2.2.0
  Downloading https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
     |████████████████████████████████| 71 kB 577 kB/s
Requirement already satisfied: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (8.1.2)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.4.1)
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (5.4.2)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.7.0)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.13)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.53.3)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5)
Requirement already satisfied: wrapt<2,>=1.10 in /usr/lib/python3/dist-packages (from deprecated->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.12.1)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5b0)
Installing collected packages: luma.core, luma.led-matrix
  Running setup.py develop for luma.led-matrix
Successfully installed luma.core-2.3.1 luma.led-matrix
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
pi@raspberrypi:~/luma.led_matrix $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from luma.led_matrix import max7219
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'luma.led_matrix'
>>>
```

So as you can see, **Successfully installed luma.core-2.3.1 luma.led-matrix** is in the output, but when I try to load a device from the module I get a 'ModuleNotFoundError'

Help!  I'm slowly losing the will to carry on with this, everything I have tried is just broken

ianglennon, 2022-01-03:
Oh, and I had to use 'sudo -H' to run pip3 because doing it as a user threw an error that directories could not be created

ianglennon, 2022-01-03:
I have cleaned everything out, removed the luma.core (which did install) and luma.led_matrix (which failed because it wasn't installed, despite the line above which says it was) and purged the pip3 cache.  This is what I did next, including the results...

```
pi@raspberrypi:~ $ git clone https://github.com/rm-hull/luma.led_matrix.git
Cloning into 'luma.led_matrix'...
remote: Enumerating objects: 2202, done.
remote: Counting objects: 100% (37/37), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 2202 (delta 10), reused 4 (delta 1), pack-reused 2165
Receiving objects: 100% (2202/2202), 11.34 MiB | 2.90 MiB/s, done.
Resolving deltas: 100% (1286/1286), done.
pi@raspberrypi:~ $ cd luma.led_matrix/
pi@raspberrypi:~/luma.led_matrix $ ls
CHANGES.rst  CONTRIBUTING.rst  doc  examples  LICENSE.rst  luma  MANIFEST.in  pyproject.toml  pytest.ini  README.rst  setup.cfg  setup.py  tests  tox.ini
pi@raspberrypi:~/luma.led_matrix $ vi setup.cfg
pi@raspberrypi:~/luma.led_matrix $ sudo pip3 install -e .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.led_matrix
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting luma.core>=2.2.0
  Downloading https://www.piwheels.org/simple/luma-core/luma.core-2.3.1-py2.py3-none-any.whl (71 kB)
     |████████████████████████████████| 71 kB 755 kB/s
Collecting rpi-ws281x
  Downloading https://www.piwheels.org/simple/rpi-ws281x/rpi_ws281x-4.3.1-cp39-cp39-linux_armv7l.whl (117 kB)
     |████████████████████████████████| 117 kB 1.9 MB/s
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (5.4.2)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.4.1)
Requirement already satisfied: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (8.1.2)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.7.0)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (0.53.3)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.13)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5)
Requirement already satisfied: wrapt<2,>=1.10 in /usr/lib/python3/dist-packages (from deprecated->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.12.1)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (3.5b0)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi->luma.core>=2.2.0->luma.led-matrix==1.5.0) (1.2.1)
Installing collected packages: rpi-ws281x, luma.core, luma.led-matrix
  Running setup.py develop for luma.led-matrix
Successfully installed luma.core-2.3.1 luma.led-matrix rpi-ws281x-4.3.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
pi@raspberrypi:~/luma.led_matrix $ cd examples/
pi@raspberrypi:~/luma.led_matrix/examples $ ./sevensegment_demo.py
Traceback (most recent call last):
  File "./sevensegment_demo.py", line 13, in <module>
    from luma.led_matrix.device import max7219
ImportError: No module named luma.led_matrix.device
```

thijstriemstra, 2022-01-03:
Let's keep this in one ticket, having all feedback in a single ticket is only way to manage this. Duplicate of #269. If we fix the error in #269 and there are still issues, I will re-open this ticket.

closed 2022-01-03 by thijstriemstra

jarleven, 2022-01-09:
> Let's keep this in one ticket, having all feedback in a single ticket is only way to manage this. Duplicate of #269. If we fix the error in #269 and there are still issues, I will re-open this ticket.

This is working for me on Bullseye. But only in a virtual environment.
https://raw.githubusercontent.com/jarleven/flaskapp/main/luma/setup_matrix.sh